RIOT-OS/RIOT #14666: Particle Xenon: add second uart_config

nicoHarel, 2020-07-31:
#### Description

Ok this is kind of a very simple feature, so I'll try to be as quick as possible. I tried to use an UART com (not the stdio one) on the Particle Xenon but noticed that the second UART configuration is missing from perif_conf.h. I thought for a second that "UART 1" wasn't implemented for the NRF52840 before looking in other periph config files from other boards using the same MCU (and also noticed this PR #10621 . 

**Is there any reason why the UART 1 isn't defined for the Particle Xenon ?  Maybe I'm missing something.**

benpicco, 2020-07-31:
> Is there any reason why the UART 1 isn't defined for the Particle Xenon?

Nobody has set it up yet.
Since the pins on the nRF52 can be mapped to pretty much any peripheral, that should be pretty easy.
Want to submit a patch?

nicoHarel, 2020-07-31:
Modified it locally and seems to work just fine. So yes, I guess I could =)

benpicco, 2020-07-31:
Nice, please do :)

closed 2021-01-04 by nicoHarel