RIOT-OS/RIOT #14667: Undefined behavior in `pkg/semtech-loramac/contrib/`

maribu, 2020-07-31:
#### Description

The C standard takes Harvard Architectures into account, for which program and data memory are distinct address spaces. This also means that they can have different address sizes, which e.g. for the AVR platform is the case: RAM addresses are always 16 bit wide and ROM addresses [can be 22 bit wide (look up the program counter size)](http://www.bound-t.com/targets/avr/index.html#description).

As a result, the C standard forbids to cast function pointers (ROM address) to data pointers (RAM addresses). (This includes that `void *` pointers cannot be used to store a function address.)

#### Steps to reproduce the issue

Check this chunks of code:

https://github.com/RIOT-OS/RIOT/blob/7e4b3d0f402c7943e3680c411025979dc013b457/pkg/semtech-loramac/contrib/semtech_loramac_timer.c#L41-L49

https://github.com/RIOT-OS/RIOT/blob/7e4b3d0f402c7943e3680c411025979dc013b457/pkg/semtech-loramac/contrib/semtech_loramac.c#L609-L615

#### Expected results

No cast from function pointer to data pointer and vice versa.

#### Actual results

A cast from function pointer to data pointer and the other direction is present.

#### Versions

Current master.

----

Discussion started in https://github.com/RIOT-OS/RIOT/pull/14663

maribu, 2020-07-31:
@fjmolinas, @aabadie, @jia200x: Care to take a look?

fjmolinas, 2020-07-31:
Would this be solved with some indirection? Passing a pointer to a function pointer?

(compiles but untested)

```diff
diff --git a/pkg/semtech-loramac/contrib/semtech_loramac.c b/pkg/semtech-loramac/contrib/semtech_loramac.c
index 97c6e5055..e18401b50 100644
--- a/pkg/semtech-loramac/contrib/semtech_loramac.c
+++ b/pkg/semtech-loramac/contrib/semtech_loramac.c
@@ -609,8 +609,8 @@ void *_semtech_loramac_event_loop(void *arg)
                 case MSG_TYPE_MAC_TIMEOUT:
                 {
                     DEBUG("[semtech-loramac] MAC timer timeout\n");
-                    void (*callback)(void) = msg.content.ptr;
-                    callback();
+                    TimerEvent_callback_t* call = msg.content.ptr;
+                    call->cb();
                     break;
                 }
                 case MSG_TYPE_LORAMAC_CMD:
diff --git a/pkg/semtech-loramac/contrib/semtech_loramac_timer.c b/pkg/semtech-loramac/contrib/semtech_loramac_timer.c
index c77b24d5b..9e288eecd 100644
--- a/pkg/semtech-loramac/contrib/semtech_loramac_timer.c
+++ b/pkg/semtech-loramac/contrib/semtech_loramac_timer.c
@@ -29,7 +29,7 @@ void TimerInit(TimerEvent_t *obj, void (*cb)(void))
 {
     obj->dev = (xtimer_t) { 0 };
     obj->running = 0;
-    obj->cb = cb;
+    obj->call.cb = cb;
 }

 void TimerReset(TimerEvent_t *obj)
@@ -44,7 +44,7 @@ void TimerStart(TimerEvent_t *obj)
     xtimer_t *timer = &(obj->dev);
     msg_t *msg = &(obj->msg);
     msg->type = MSG_TYPE_MAC_TIMEOUT;
-    msg->content.ptr = obj->cb;
+    msg->content.ptr = &obj->call;
     xtimer_set_msg(timer, obj->timeout, msg, semtech_loramac_pid);
 }

diff --git a/pkg/semtech-loramac/include/semtech-loramac/timer.h b/pkg/semtech-loramac/include/semtech-loramac/timer.h
index cce20f208..c3daddebd 100644
--- a/pkg/semtech-loramac/include/semtech-loramac/timer.h
+++ b/pkg/semtech-loramac/include/semtech-loramac/timer.h
@@ -31,6 +31,11 @@ extern "C" {

 #include "semtech_loramac.h"

+typedef struct TimerEvent_callback
+{
+   void (*cb)(void);
+} TimerEvent_callback_t;
+
 /**
  * @brief   Timer object description
  */
@@ -39,7 +44,7 @@ typedef struct TimerEvent_s {
     uint8_t running;    /**< Check if timer is running */
     xtimer_t dev;       /**< xtimer instance attached to this LoRaMAC timer */
     msg_t msg;          /**< message attacher to this LoRaMAC timer */
-    void (*cb)(void);   /**< callback to call when timer timeout */
+    TimerEvent_callback_t call;   /**< callback to call when timer timeout */
 } TimerEvent_t;

 /**
```

fjmolinas, 2020-07-31:
> Would this be solved with some indirection? Passing a pointer to a function pointer?

Asking because I'm not sure I understand the issue completely.

maribu, 2020-07-31:
Yes, that would solve the issue. But it seems that just using the `uint32_t` value works; Murdock seemed not to complain. And I think it is pretty safe to assume that `uint32_t` is wide enough for function pointers on constrained devices. So @benpicco already solved the issue in https://github.com/RIOT-OS/RIOT/pull/14663

benpicco, 2020-07-31:
#14663 is not merged yet :wink:

maribu, 2020-07-31:
> #14663 is not merged yet :wink:

I could fix that ;-)

PR #14663 (2020-07-31): treewide: compile fixes for native

### Contribution description

`char` is unsigned on ARM but signed on x86.
This lead to a few compile errors when adding the `periph_gpio` feature to `native`, because now code that was previously never build for x86 got hit by this difference.
The host compiler is also usually newer than the embedded one, which usually means it is stricter.

This makes offending `chars` explicitly `unsigned` and works around some pedantry.

### Testing procedure

No logic changes. In fact, the binaries should stay the same.

### Issues/PRs references

split off & needed for [... cut]
closed 2020-07-31 by maribu