RIOT-OS/RIOT #14689: dist/tools/sliptty/start_network.sh: IPv6 connectivity is broken on PC

trickeydan, 2020-08-03:
#### Description

`dist/tools/sliptty/start_network.sh` [enables forwarding on all interfaces](https://github.com/RIOT-OS/RIOT/blob/7f1f8ae576c21fb4911975818a6fb6d3c47204cd/dist/tools/sliptty/start_network.sh#L37) when running.

By default, Linux will ignore RAs on interfaces with forwarding enabled unless `net.ipv6.conf.<interface>.accept_ra=2`. This is sensible for security on routers.

It thus ignores RAs on my uplink interface and breaks connectivity.

Interestingly, this seems to be partially acknowledged in the script, as it sets the appropriate sysctl for the `tun` interface. See [here](https://github.com/RIOT-OS/RIOT/blob/7f1f8ae576c21fb4911975818a6fb6d3c47204cd/dist/tools/sliptty/start_network.sh#L42)

Arguably, enabling forwarding on all interfaces is very dangerous and potentially a security issue for users that are unfamiliar with Linux routing.

I am unsure of the best approach to take to fix this, as it is not clear why forwarding is activated for all interfaces!

benemorius, 2020-08-03:
The reason that forwarding is activated for all interfaces is that [confusingly](https://unix.stackexchange.com/a/348534) the Linux kernel doesn't allow enabling IPv6 routing for independent interfaces. To get any forwarding at all, `net.ipv6.conf.all.forwarding` must be set.

The `accept_ra=2` workaround you've done is indeed the intended way to get your RAs back on the specific interface you need it from.

It's a hard problem to solve in a script because some knowledge about the network topology (which interface is the uplink?) is required. Suggestions are certainly welcome.

trickeydan, 2020-08-03:
Surely this script should work in line with all of the other `start_network.sh`? Those only enable forwarding on the BR interface, giving choice of forwarding to the user.

On systems with a poorly configured firewall, this script poses a serious risk in it's current state. It also breaks access to a fair amount of the internet. 

Related to the inconsistency between scripts: #14618

miri64, 2020-08-03:
> Surely this script should work in line with all of the other `start_network.sh`? Those only enable forwarding on the BR interface, giving choice of forwarding to the user.

Then the other scripts are wrong at least for Linux. @benemorius is right (and the linked doc backs him).  While the variable exists for single interfaces, on Linux only setting `net.ipv6.conf.all.forwarding` is the only way to get forwarding enabled (you can check with `radvd` which will fail to start without forwarding enabled).

trickeydan, 2020-08-04:
> Then the other scripts are wrong at least for Linux.

As it stands, changing this will break IPv6 connectivity on the other scripts too. As about [a third](https://www.google.com/intl/en/ipv6/statistics.html) of users have IPv6 these days, this will likely cause problems for more people.

Additionally, the other scripts use Layer 2 ethernet interfaces (`tap` for ethos, `eth` for cdc-ecm) which **do not require forwarding to be enabled** to reach nodes. If forwarding was required for these scripts, it would probably have been noticed by now.

Enabling forwarding on all interfaces on these scripts is likely to break things for a number of users, including myself and others doing work on RIOT at my University. I think it would be a massive shame to spread this issue to other scripts.

Side note: Theoretically you could enable forwarding for all interfaces, and set `accept_ra=2` for the uplink(s), but it would be very difficult to determine and account for every network topology in use. Unfortunately we can't even look at interfaces with default routes, as there may be multiple default routes, and smaller routes down other interfaces too.

miri64, 2020-08-04:
See https://github.com/RIOT-OS/RIOT/pull/10477#discussion_r241082394.

miri64, 2020-08-04:
> Additionally, the other scripts use Layer 2 ethernet interfaces (`tap` for ethos, `eth` for cdc-ecm) which **do not require forwarding to be enabled** to reach nodes. If forwarding was required for these scripts, it would probably have been noticed by now.

You could just configure the forwarding according to the uplink.

trickeydan, 2020-08-04:
> You could just configure the forwarding according to the uplink.

My main concern with this is that often the first experience of RIOT that someone has is using one of the example programs, e.g `examples/gnrc_border_router`. If the user does not have knowledge of how forwarding works on Linux, all they will notice is that their connectivity has broken. If these changes are made, then it would need to be very well documented. This documentation should include the security risk that comes with enabling IP forwarding on systems with a firewall that is not configured for it (i.e desktop distros).

It would be a massive shame if the examples were no longer as simple to use as `make all flash term`.

miri64, 2020-08-04:
>  If the user does not have knowledge of how forwarding works on Linux, all they will notice is that their connectivity has broken.

I don't experience any broken connectivity when using this script, so I am not clear on what you mean by that. Please clarify.

trickeydan, 2020-08-04:
> I don't experience any broken connectivity when using this script, so I am not clear on what you mean by that. Please clarify.

As mentioned previously by myself further up in the issue, when IPv6 forwarding is enabled on an interface, Linux will by default ignore RAs on that interface. This means that IPv6 connectivity to the wider internet is lost when forwarding is enabled and no further action is taken (i.e `accept_ra=2`). As about a third of users now use IPv6 (and some internet services are IPv6-only), not having IPv6 connectivity is an issue for many.

I noticed this issue as my SSH connection to a server dropped when this script was ran. My laptop had dropped the IPv6 default route from my router as it was ignoring RAs, later I lost the IPv6 address too (also due to lack of RA).

miri64, 2020-08-04:
Ah sorry, I did not read carefully enough :confused:. Fact is, for a border router (what this script effectively wants to enable) forwarding capabilities are required, especially if one wants to use it as e.g. @benpicco does in https://github.com/RIOT-OS/RIOT/issues/14676 (as said: RAdvD requires forwarding to be enabled, not sure if it is happy with "enabling" forwarding only on the operating interface is enough). As in my desired fix for #14618 everything those script do could be configured via CLI parameter how about this: only enable forwarding when a `-f` option is provided + setting `accept_ra=2` for _all_ interfaces (`for iface in $(ip link) ...`) in that case.

trickeydan, 2020-08-04:
> only enable forwarding when a -f option is provided + setting accept_ra=2 for all interfaces (for iface in $(ip link) ...)

This sounds like a good solution to me. :+1:

It could potentially be nice to warn if a layer 3 interface (SLIP or similar) is used without the -f option being given. This would hopefully allow users to be able to work out why packets aren't being forwarded across the interface, although that is more in scope of #14618.

benemorius, 2020-09-11:
@miri64 do you have any near plans to address this? It's not been high on my list but I can do it. I think your proposed solution is fine, and better then current master.

miri64, 2020-09-11:
@benemorius I work on it whenever I find some time. First step (#14881) is already merged.

maribu, 2023-05-18:
@miri64: With https://github.com/RIOT-OS/RIOT/pull/14928 and https://github.com/RIOT-OS/RIOT/pull/15087 merged, can this be considered fixed?