RIOT-OS/RIOT #14750: periph_timer test doesn't work with the waspmote-pro board

jdavid, 2020-08-12:
#### Description

`tests/periph_timer` and `tests/periph_rtt` don't work with the **waspmote-pro** board

ping @kYc0o @ArcticSnow

#### Steps to reproduce the issue

Tested with the master branch of RIOT:

```bash
$ cd tests/periph_timer
$ make BOARD=waspmote-pro PORT=/dev/ttyUSB0 FFLAGS_EXTRA="-F" all flash
[...]
$ make BOARD=waspmote-pro BOOTLOADER_BAUD=115200 PORT=/dev/ttyUSB0 term
[...]
2020-08-12 16:34:06,601 # Help: Press s to start test, r to print it is ready
s
2020-08-12 16:34:16,839 # START
2020-08-12 16:34:16,903 # main(): This is RIOT! (Version: 2020.10-devel-699-g02a71)
2020-08-12 16:34:16,903 # 
2020-08-12 16:34:16,934 # Test for peripheral TIMERs
2020-08-12 16:34:16,935 # 
2020-08-12 16:34:16,951 # Available timers: 2
2020-08-12 16:34:16,951 # 
2020-08-12 16:34:16,967 # Testing TIMER_0:
2020-08-12 16:34:17,014 # TIMER_0: ERROR on initialization - skipping
2020-08-12 16:34:17,015 # 
2020-08-12 16:34:17,015 # 
2020-08-12 16:34:17,046 # Testing TIMER_1:
2020-08-12 16:34:17,079 # TIMER_1: ERROR on initialization - skipping
2020-08-12 16:34:17,079 # 
2020-08-12 16:34:17,094 # 
2020-08-12 16:34:17,095 # TEST FAILED
```

Regarding `periph_rtt` I get a build error, tested with the master branch, the 2020.07 release and older releases:

```bash
$ cd tests/periph_rtt
$ make BOARD=waspmote-pro all
There are unsatisfied feature requirements: periph_rtt
/home/jdavid/sandboxes/UiO/wsn_riot/RIOT/tests/periph_rtt/../../Makefile.include:895: *** You can let the build continue on expected errors by setting CONTINUE_ON_EXPECTED_ERRORS=1 to the command line.  Stop.
```

#### Expected results

Expected it to work, as for example it does with the **remote-revb** board:

```bash
$ cd tests/periph_timer
$ make BOARD=remote-revb all flash term
[...]
2020-08-12 17:19:39,937 # Help: Press s to start test, r to print it is ready
s
2020-08-12 17:19:48,923 # START
2020-08-12 17:19:48,928 # main(): This is RIOT! (Version: 2020.10-devel-699-g02a71)
2020-08-12 17:19:48,928 # 
2020-08-12 17:19:48,929 # Test for peripheral TIMERs
2020-08-12 17:19:48,929 # 
2020-08-12 17:19:48,930 # Available timers: 4
2020-08-12 17:19:48,930 # 
2020-08-12 17:19:48,931 # Testing TIMER_0:
2020-08-12 17:19:48,932 # TIMER_0: initialization successful
2020-08-12 17:19:48,932 # TIMER_0: stopped
2020-08-12 17:19:48,933 # TIMER_0: set channel 0 to 5000
2020-08-12 17:19:48,933 # TIMER_0: set channel 1 to 10000
2020-08-12 17:19:48,934 # TIMER_0: starting
2020-08-12 17:19:48,937 # TIMER_0: channel 0 fired at SW count      263 - init:      263
2020-08-12 17:19:48,949 # TIMER_0: channel 1 fired at SW count      551 - diff:      288
2020-08-12 17:19:48,950 # 
2020-08-12 17:19:48,950 # Testing TIMER_1:
2020-08-12 17:19:48,951 # TIMER_1: initialization successful
2020-08-12 17:19:48,951 # TIMER_1: stopped
2020-08-12 17:19:48,952 # TIMER_1: set channel 0 to 5000
2020-08-12 17:19:48,955 # TIMER_1: set channel 1 to 10000
2020-08-12 17:19:48,956 # TIMER_1: starting
2020-08-12 17:19:48,963 # TIMER_1: channel 0 fired at SW count      263 - init:      263
2020-08-12 17:19:48,974 # TIMER_1: channel 1 fired at SW count      551 - diff:      288
2020-08-12 17:19:48,974 # 
2020-08-12 17:19:48,974 # Testing TIMER_2:
2020-08-12 17:19:48,975 # TIMER_2: initialization successful
2020-08-12 17:19:48,976 # TIMER_2: stopped
2020-08-12 17:19:48,977 # TIMER_2: set channel 0 to 5000
2020-08-12 17:19:48,978 # TIMER_2: starting
2020-08-12 17:19:48,990 # TIMER_2: channel 0 fired at SW count      277 - init:      277
2020-08-12 17:19:48,990 # 
2020-08-12 17:19:48,990 # Testing TIMER_3:
2020-08-12 17:19:48,991 # TIMER_3: initialization successful
2020-08-12 17:19:48,992 # TIMER_3: stopped
2020-08-12 17:19:48,993 # TIMER_3: set channel 0 to 5000
2020-08-12 17:19:48,994 # TIMER_3: starting
2020-08-12 17:19:49,003 # TIMER_3: channel 0 fired at SW count      277 - init:      277
2020-08-12 17:19:49,004 # 
2020-08-12 17:19:49,005 # TEST SUCCEEDED
```

The `periph_rtt` test works as well with the **remote-revb** board.

#### Actual results

See *Steps to reproduce* above.

#### Versions

```
$ make print-versions

Operating System Environment
----------------------------
         Operating System: Gentoo 
                   Kernel: Linux 5.4.48-gentoo x86_64 Intel(R) Core(TM) i3-5010U CPU @ 2.10GHz
             System shell: GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Gentoo 9.3.0-r1 p3) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
                  avr-gcc: avr-gcc (Gentoo 9.3.0-r1 p3) 9.3.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 10.0.0 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.16.5
                 cppcheck: Cppcheck 1.88
                  doxygen: missing
                      git: git version 2.26.2
                     make: GNU Make 4.2.1
                  openocd: missing
                   python: Python 3.7.8
                  python2: Python 2.7.18
                  python3: Python 3.7.8
                   flake8: 3.8.1 (mccabe: 0.6.1, pycodestyle: 2.6.0, pyflakes: 2.2.0) CPython 3.7.8 on
               coccinelle: missing
```

benpicco, 2020-08-14:
For RTT support the board needs a 32 kHz crystal. Does the board have one?
You can try with `CONTINUE_ON_EXPECTED_ERRORS=1` and see if it works.

Normal timers should work though.
@maribu that ATmega1281 runs at 14.7456 MHz, is that at odds with what the timer implementation expects?

maribu, 2020-08-14:
No, the issue with the `waspmote-pro` is that the timer configuration is completely broken. The timer frequency needs to be generated from the core clock using a prescaler. But no prescaler is able to generate the frequency configured by the board.

The fix is also not that simple: `xtimer` a frequency that complies with one of the following:

1. The frequency is 1 MHz
2. Efficient frequency conversion to 1 MHz can be performed by binary shifts
3. The frequency is 32.768 kHz

None of the `waspmote-pro` timers is compatible with xtimer. Instead, one would have to first fix the timer configuration and then use `ztimer`. `xtimer_on_ztimer` would also work, but currently the check for applicable frequencies is not disabled with `xtimer_on_ztimer`. It should be too difficult to fix the issue with `xtimer_on_ztimer`.

I saw the issue in the `waspmote-pro` when I wanted to get my custom ATmega1284P breakout board clocked at 12 MHz working. But I assumed that as the timer configuration of the waspmote was broken since it got merged, no one is possibly using it. So my plan was to create a PR dropping that board, but I didn't managed to create it.

But now that there is someone apparently with access to this hardware and interest in the board, we might as well just fix the issues :-)

jdavid, 2020-08-18:
@maribu 

> [...] first fix the timer configuration [...]

I'm new to RIOT. So ztimer can work over `periph_timer` or `periph_rtt`, right?

From the MCU datasheet:

- Two 8-bit Timer/Counters with Separate Prescaler and Compare Mode
- Four 16-bit Timer/Counter with Separate Prescaler, Compare- and Capture Mode
- Real Time Counter with Separate Oscillator

There's no reference to a RTT in the datasheet, so it should be `periph_timer`, right?

maribu, 2020-08-18:
> I'm new to RIOT.

Welcome :-)

> So `ztimer` can work over periph_timer or periph_rtt, right?

Yes. It can actually do both and use `periph_timer` for short durations and high resolutions and `periph_rtt` for long sleeps. This allows the board to do power saving when no soft timer is active on top of `periph_timer`. (However, power saving is not yet implemented on ATmega :-/)

RTT is RIOT-speak for a Real Time Counter (RTC) which counts in one-dimensional ticks in order to distinguish between Real Time Clocks (RTC) that counts in years, months, days, hours, minutes, seconds. The latter is referred to as RTC in RIOT. For the AVR platform, both interfaces are implemented on top of the same hardware - one can therefore only use one.

If you have hardware to test, I might be able to create a pull request to provide a working clock configuration this week.

jdavid, 2020-08-18:
Thanks! Yes I've the hardware to test, waspmote boards v12 and v15. Both have the same MCU ATmega1281 (the main difference is the external RTC).

maribu, 2020-08-18:
> (the main difference is the external RTC).

Could you share a link to some documentation? The difference in the external RTC sounds like it could be pretty relevant for the clock configuration :-)

jdavid, 2020-08-19:
v12 RTC is DS3231SN, see http://www.libelium.com/downloads/documentation/v12/waspmote_technical_guide.pdf

v15 RTC is DS1337C, see http://www.libelium.com/downloads/documentation/waspmote_technical_guide.pdf

maribu, 2020-08-19:
Thanks for sharing. As those are external RTCs, those require dedicated drivers. There is a driver for the DS1307 already in RIOT; I would guess that this driver would work out of the box for the DS1337C or at least with minor modifications for both clocks.

The ATmega1284p does btw. have an internal RTC, it just needs an 32.678 kHz crystal (and some passive components). Maybe the external RTC has a lower power consumption.

jdavid, 2020-08-19:
Thanks, that was fast!

However the tests `tests/periph_timer` and `tests/periph_rtt` still fail as described in this issue.

After reading PR #14783 I tried as well:

    USEMODULE=xtimer_on_ztimer BOARD=waspmote-pro make -C tests/xtimer_usleep

It builds, but the program hangs when running in the board.

If I understand correctly PR #14783 is a partial fix and this issue should be re-opened?

maribu, 2020-08-19:
> If I understand correctly PR #14783 is a partial fix and this issue should be re-opened?

Exactly. It was auto-closed because the PR contained the word "fix" and a reference to this issue. Sadly, that grammar is not quite accurate.

jdavid, 2020-08-20:
`tests/periph_timer` still doesn't work, with the same output

closed 2020-09-24 by fjmolinas