RIOT-OS/RIOT #14752: Return value of gnrc_netif_ipv6_addr_idx is not checked for negative

keestux, 2020-08-13:
#### Description
The return value of `gnrc_netif_ipv6_addr_idx` is not always checked for negative values before it is used as an index in an array.

For example in `nib.c`
```
        int idx = gnrc_netif_ipv6_addr_idx(tgt_netif, &nbr_sol->tgt);

        if (gnrc_netif_ipv6_addr_dad_trans(tgt_netif, idx)) {
```

BTW This is one of the findings of Coverity Scan. And I'm searching for a nice method to report findings like this. It may not always be that it is a real issue. Some can be obvious, some need discussion.

miri64, 2020-08-28:
TL;DR checks should be put in place in addition to some locking.

I identified four instances of where the index is used on the address array without checking first:

#### `nib.c`
(in both instances: `gnrc_netif_ipv6_addr_dad_trans()` accesses array without checking the provided index first)
##### `_handle_nbr_sol()`
https://github.com/RIOT-OS/RIOT/blob/7eed33bf7870a5e535d60f217acd70ac9cc05c0f/sys/net/gnrc/network_layer/ipv6/nib/nib.c#L916-L918

##### `_handle_nbr_adv()`
https://github.com/RIOT-OS/RIOT/blob/7eed33bf7870a5e535d60f217acd70ac9cc05c0f/sys/net/gnrc/network_layer/ipv6/nib/nib.c#L1048-L1050

#### `_nib-6ln.c`
##### `_handle_aro()`
https://github.com/RIOT-OS/RIOT/blob/7eed33bf7870a5e535d60f217acd70ac9cc05c0f/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c#L98-L107

##### `_handle_rereg_address()`
(`_is_valid()` accesses the array without checking)
https://github.com/RIOT-OS/RIOT/blob/7eed33bf7870a5e535d60f217acd70ac9cc05c0f/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c#L198-L201

---

For the one in `nib.c` and `_handle_rereg_address()` in `_nib-6ln.c` the check is technically not necessary, as the interface in these three cases is gotten based on the address we get the interface. However, it could be argued, that between those lines that the address is removed for some reason, so maybe a check is necessary. In any case the functions should acquire exclusive access to the interface to ensure they really work on the address in question.

For `_handle_aro()` the check is definitely needed, because while the `netif` is also determined from `ipv6->dst` in `gnrc_ipv6.c`'s `_pkt_not_for_me()` there is a "long" time between that and the NIB acquiring exclusive access to `netif` in `gnrc_ipv6_nib_handle_pkt()`.

I will open a PR to fix all of that.

miri64, 2020-08-28:
See #14883.

commit f9c3e5f5 (2020-09-07): gnrc_ipv6_nib: check return value of gnrc_netif_ipv6_addr_idx()

And acquire exclusive network interface access when necessary.

Fixes #14752.
closed 2020-09-07 by benpicco