RIOT-OS/RIOT #14771: cpu/esp*: Access to scheduler internals

maribu, 2020-08-17:
#### Description

The ESP32 and ESP8266 platform specific code is accessing scheduler internal stuff and writing to it. E.g. `NORETURN void task_exit(void)` is with `__XTENSA_CALL0_ABI__` indeed just mapping to `NORETURN void sched_task_exit(void)`, but otherwise is handling the task exit internally.

IMO, we should make sure that platform specific code is not modifying the scheduler internal state. Instead, a well defined API should be used for interactions between platform specific code and the scheduler. This would allow changing the internals of the scheduler without breaking other stuff

#### Suggestion

##### Use existing scheduler API, such as `sched_task_exit()` 

Replace code writing to `sched_*` by calls to the existing scheduler API. E.g. rewrite the `task_exit()` function to also use `sched_task_exit()` even when `__XTENSA_CALL0_ABI__` is not used. (Btw: Is there indeed a need to support two different ABIs?)

Or would this be infeasible to get the ESP platform working on top of the existing core APIs?

#### Extend the scheduler API

If indeed the current core APIs are not enough to get the ESP platform spinning, we should IMO just extend the APIs as needed and use that instead.

#### Versions

Current master

maribu, 2020-08-17:
`vTaskDelete()` is also anther function reinventing the `sched_task_exit()` wheel.

gschorcht, 2020-08-17:
The code is a bit trickle, but it was the only way to get it working with the existing API for ESPs. 

> Btw: Is there indeed a need to support two different ABIs?)

Yes, although ESPs are Xtensa cores, the ESP8266 uses the Call0 ABI while ESP32 uses the Window API. That's why ESP32 requires special handling on `task_exit`.

maribu, 2020-08-17:
The first *n* lines of the ESP32 `task_exit()` and `sched_task_exit()` do exactly the same: Updating the scheduler internals. Afterwards, `sched_task_exit()` calls into the cpu code, namely `cpu_switch_context_exit()`, while `task_exit()` does other stuff.

Why wouldn't it be possible to just use `sched_task_exit()` and move the other stuff from `task_exit()` to `cpu_switch_context_exit()`?

gschorcht, 2020-08-17:
It might work. To be honest, I don't remember exactly why I did it in that way. I only remember that applications crashed that return from the main thread.