RIOT-OS/RIOT #14792: [tracker] ieee802154_hal: implementation status

jia200x, 2020-08-19:
## Description

This tracker list the on-going efforts to implement the IEEE 802.15.4 Radio HAL (#13943) for the existing IEEE 802.15.4 radios, since #14371 got merged.

The `netdev` interface should still be available for each radio if `MODULE_IEEE802154_RADIO_HAL` is not present. This way we can have both interface and ease the migration process. Also, each implementation should adapt the `ieee802154_hal` test (provided in #14791) to test the implementation.

Use #14791 as a reference.

## List of HAL implementations

- [ ] CC2538 (#14791 ). It also includes the test application and `ieee802154_radio_hal` pseudomodule.
- [ ] at86rf2xx
- [ ] at86rf215
- [ ] CC2420
- [ ] nrf52840
- [ ] kw41z
- [ ] kw2xrf
- [ ] mrf24j40

jia200x, 2020-08-20:
@benpicco do you think you could take care of the AT86RF215?

jia200x, 2020-10-16:
I will personally write the `at86rf2xx` HAL implementation, since it was one of the references for the HAL design. Any volunteers for the others? :)

benpicco, 2020-10-16:
at86rf215 is on my list, but I don't have an ETA for it.
I can also take a look at mrf24j40, being a full-MAC device that should be easy, right?

jeandudey, 2020-10-16:
>Any volunteers for the others? :)

`cc26x2_cc13x2_rf` is in the on-going list :-)

jia200x, 2020-10-19:
> at86rf215 is on my list, but I don't have an ETA for it.

Great!

> I can also take a look at mrf24j40, being a full-MAC device that should be easy, right?

Since the mrf24j40 handles TX ACK, it should be easy. It should be as difficult as the cc2538, which was not too bad.

jia200x, 2021-06-07:
:eyes: :eyes: :eyes:  :eyes:  :eyes:  :eyes:

jia200x, 2022-08-19:
updated list to current porting status

jia200x, 2026-09-08:
@RIOT-OS/maintainers 

@Stopkaa was able to finish (almost) all remaining PRs. The only one missing is the CC2420.

Regarding that one, I propose to exclude it from the migration process and either deprecate it (hasn't been maintained) or leave it with the old `netdev` interface.

My two cents:
- It's a 22 years old radio that was designed around the old IEEE 802.15.4 2003 version. It doesn't have features that are relevant for many applications (TX_DONE interrupt, etc) and the hardware is hard to get nowadays. Last commit touching radio functionality was 6 years ago (d8794d7c5ac0d74b79a4ecd570d8c5be634d6ac1) and it hasn't been touched anymore, besides chores.
- Although the Radio HAL was designed around the use case of CC2420, any practical implementation of a SubMAC layer on top would require polling, which introduces unnecessary complexity for the upper layer and it's probably quite inefficient.

I'm personally in favor of deprecating it. What do you think?
If this is clarified, I think we can close the issue and consider the Radio HAL migration complete.

Teufelchen1, 2026-09-09:
>  The only one missing is the CC2420. [..] .. the hardware is hard to get nowadays. 

Just to clarify, this is the radio we are talking about? Because it is still listed as "active" and available to be bought. https://www.ti.com/product/CC2420

jia200x, 2026-09-10:
> Just to clarify, this is the radio we are talking about? Because it is still listed as "active" and available to be bought. https://www.ti.com/product/CC2420

[...Mostly out of stock...](https://www.digikey.com) ;).
On the other hand, while the SoC is still being produced, the two CC2420 boards in RIOT (i.e., Zolertia Z1 and Telos-B) can only be found on Ebay and there doesn't seem to be an active community behind them.

As it is, it's only supported by very basic variants of GNRC and does not allow for any MAC features.
Additionally, this is pushing back the radio rework, as we cannot really cut the current gnrc_netif path without integrating the Radio HAL directly. IMO we shouldn't let this radio block the ongoing efforts.

If someone is willing to port it, please speak up. Otherwise, neither @Stopkaa nor me are willing to spend time on it at the moment, as there are still things to be done for the migration.

Teufelchen1, 2026-09-11:
Sure, I agree that we can drop it.

jia200x, 2026-09-11:
ok, let's merge the two remaining PRs and I will close the issue.

crasbe, 2026-09-11:
> ok, let's merge the two remaining PRs and I will close the issue.

I'd add "deprecate C2420" to the list before closing this. Perhaps we should think about deprecating the Telos-B and Zolertia Z1 at the same time.

crasbe, 2026-09-16:
I created a deprecation PR for the `cc2420` driver and the `z1` and `telosb` boards: #22716

miri64, 2026-09-17:
> I created a deprecation PR for the `cc2420` driver and the `z1` and `telosb` boards: [#22716](https://github.com/RIOT-OS/RIOT/pull/22716)

Do we have an alternative `z1` board? It used to be my testing board for MSP430 MCUs for the longest time.

maribu, 2026-09-17:
The plan for the z1 is to deprecate it without removal timeline and find another MSP430 board with at least a similar amount of RAM and flash that can still be sourced easily.

TI still produces MSP430 dev boards and Olimex as 3rd party vendor also actively produces MSP430 boards. I will look into porting one of those as replacement prior to removal of the z1.

miri64, 2026-09-17:
Great, thanks @maribu 🙏