RIOT-OS/RIOT #14855: RIOT uses old configuration file for ST-link download via OpenOCD.

jhaand, 2020-08-26:
#### Description
When downloading code to the Bluepill board via OpenOCD, I get a mention that the currently used stlink-v2-1.cfg is deprecated. 
```WARNING: interface/stlink-v2-1.cfg is deprecated, please switch to interface/stlink.cfg```

#### Steps to reproduce the issue
Download code to a bluepill board using STlink and openocd. Using `make flash`.

#### Expected results
No warnings and that OpenOCD uses the correct configuration file. 

#### Actual results
I get the following warning by OpenOCD when flashing the board. 
```WARNING: interface/stlink-v2-1.cfg is deprecated, please switch to interface/stlink.cfg```
According to the OpenOCD repository, this change has been introduced in 2017. 
https://sourceforge.net/p/openocd/code/ci/31c58c139d85c35cc8ebce4196edb2c5eb157c7a/tree//tcl/interface/stlink-v2-1.cfg?diff=7568a91c8e2398a113f0b40a2a24a1b91ed12c95

#### Versions
```
Operating System Environment
----------------------------
         Operating System: "Debian GNU/Linux" 
                   Kernel: Linux 5.7.0-2-amd64 x86_64 unknown
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Debian 10.1.0-6) 10.1.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (15:8-2019-q3-1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 9.0.1-13 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.16.3
                 cppcheck: Cppcheck 2.1
                  doxygen: missing
                      git: git version 2.28.0
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.10.0+dev-snapshot (2020-08-19-07:12)
                   python: Python 2.7.18
                  python2: Python 2.7.18
                  python3: Python 3.8.5
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

fjmolinas, 2020-08-26:
The reason why we have kept this so far was mentioned in https://github.com/RIOT-OS/RIOT/pull/11824#issuecomment-541647434. Although it is deprecated there has not been a new OpenOCD release since AFAIK. Has this changed?

fjmolinas, 2020-08-26:
@jhaand maybe something like suggested in that #11824 could be implemented?

jhaand, 2020-08-26:
It hasn't changed in the mean time. 
I wrote a ticket in the OpenOCD project to address this issue. I think this will need to be fixed where it's appropriate. And currently that responsibility lies with the OpenOCD project. 

link to the ticket: 
https://sourceforge.net/p/openocd/tickets/277/

jhaand, 2020-08-26:
@fjmolinas: The solution mentioned in #11824 looks good. I will try that and see if it works. 
At least that would make life a lot clearer for users.

jhaand, 2020-08-26:
@fjmolinas Update has been pushed as suggested in the solution for #11824 
The makefile now checks for stlink.cfg or stlink-v(VERSION).config

It works with both the old and new versions of OpenOCD. 
I wasn't able to break the lines in the Makefile, so it has become a bit long.

PR #14856 (2020-09-01): Updated stlink makefile to use OpenOCD's single configuration file.

### Contribution description
OpenOCD has been using a single configuration file for all ST-link adapters. RIOT still used a specific version. Which leads to a warning when flashing a board. This patch makes RIOT use the generic stlink.cfg which OpenOCD has been using since 2017.

### Testing procedure
Flash firmware to a board via an ST-Link adapter using OpenOCD. I tested this with a Bluepill board. 

### Issues/PRs references
This Fixes #14855
closed 2020-09-01 by fjmolinas