RIOT-OS/RIOT #14894: stm32f405rg support

Jalili-73, 2020-08-31:
#### Description
how can i add personal board with stm32f405rg mcu or witch cpu work in stm32f405rg.

fjmolinas, 2020-09-01:
Hi @Jalili-73, `stm32f4` family is supported in RIOT so it should be pretty simple to add support for a BOARD using `stm32f405rg`.

First I would recommend you take a look at http://riot-os.org/api/porting-boards.html, it will give you a general idea of how `BOARD`s directories are structured. Consider using the `make generate-board` target, documented in the previous link. This will give you the basic frame for adding your `BOARD`.

Then you should start filling up the empty files, it can be a good starting point to look at a similar `BOARD`. In your case this could be the [`pyboard`](https://github.com/RIOT-OS/RIOT/tree/master/boards/pyboard) since it has the same `CPU` so a lot of the features might be similar. Is your `BOARD`a custom one? If it is not if you tell me which one it is maybe we already have one in RIOT with a similar layout, and I kind point you to it.

Let me know if this information is enough.

Jalili-73, 2020-09-01:
> Hi @Jalili-73, `stm32f4` family is supported in RIOT so it should be pretty simple to add support for a BOARD using `stm32f405rg`.
>
> First I would recommend you take a look at [...]

Hi fjmolinas
I really appreciate your help.
I'm beginner in RIOT os. I studied riot course in github but this didn't satisfy me. 
can you guide me to learn RIOT?
there is a board similar to stm32f4discovery with MCU:stm32f405rg i want to use RIOT on this board.

fjmolinas, 2020-09-01:
>  I'm beginner in RIOT os. I studied riot course in github but this didn't satisfy me.

In what way was it not satisfying, what where you expecting? what where your issues?

Jalili-73, 2020-09-02:
> > I'm beginner in RIOT os. I studied riot course in github but this didn't satisfy me.
> 
> In what way was it not satisfying, what where you expecting? what where your issues

I'm working on electronic board that must include very feature. reading data from sensors and GPS using UART/USART and i2c processing data do something in different situation and finally send data with sim800 to server.
I'm going to use RIOT to power on board.
I read RIOT course and riot-os.org but it not very clear for me.
I'm going to learn RIOT for doing this project. my challenge is:
- create new board and config board,
- write some application on RIOT.

as you know the speed of learning dependent on resource that you study. i have experience on c programming and know the role of operating system.
I'm searching for way of RIOT learning that step by step guide me to learn and resolve my issues.

fjmolinas, 2020-09-03:
@Jalili-73 Is this the course you followed? https://github.com/RIOT-OS/riot-course

fjmolinas, 2020-09-03:
> create new board and config board,

This should have been answered by my initial post.

Jalili-73, 2020-09-05:
> @Jalili-73 Is this the course you followed? https://github.com/RIOT-OS/riot-course

yes

Jalili-73, 2020-09-05:
> answered

closed 2020-09-05 by Jalili-73

Jalili-73, 2020-09-05:
> > create new board and config board,
> 
> This should have been answered by my initial post.

Yes that's right.
Many Thanks for your help my friend.

Jalili-73, 2020-09-06:
> > create new board and config board,
> 
> This should have been answered by my initial post.
how may i have direct communication with you for some question; whats app or email or skype or any things else