RIOT-OS/RIOT #14957: Return value of gnrc* calls not checked for -1

keestux, 2020-09-06:
#### Description
The return value of a few gnrc_* functions are used as an index in an array. These functions can return -1, but not all callers check for the -1.

The following are identified by Coverity Scan.

- `sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c:999`
- `sys/net/gnrc/network_layer/ipv6/nib/nib.c:919`
- `sys/net/gnrc/network_layer/ipv6/nib/nib.c:1051`

miri64, 2020-09-07:
>     * sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c:999

While this is error class to what is described in #14883, I think we have bigger problems when the DODAG ID of the root somehow disappears in that short amount of time from the interface (`netif` is gotten from the very same address a few lines above.

miri64, 2020-09-07:
> * `sys/net/gnrc/network_layer/ipv6/nib/nib.c:919`
> 
> * `sys/net/gnrc/network_layer/ipv6/nib/nib.c:1051`
> 
> * `/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c:198`

Fixed in #14883 and duplicate report from #14752

miri64, 2020-09-07:
> While this is error class to what is described in #14883, I think we have bigger problems when the DODAG ID of the root somehow disappears in that short amount of time from the interface (`netif` is gotten from the very same address a few lines above.

See https://github.com/RIOT-OS/RIOT/pull/14961

closed 2020-09-07 by benpicco

keestux, 2020-09-07:
Sorry for the duplicate. I had totally forgotten that I already created #14961