RIOT-OS/RIOT #14980: hifive1b: sys/ztimer/core.c:159 => *** RIOT kernel panic: FAILED ASSERTION.

mrusme, 2020-09-08:
#### Description

I'm trying to port a code that runs perfectly fine on the `nucleo-l432kc` onto the `hifive1b`. The code compiles successfully and is being flashed onto the HiFive1 revB. However, it seems like there's something wrong with the xtimer (at least that's what my code is using) and the underlying(?) ztimer implementation on the hifive. I'm not sure if this might have something to do with #13109.

```
Bench Clock Reset Complete

ATE0--> Send Flag error: #0 #0 #0 #0 AT+BLEINIT=0-->AT+BLEINIT=0
OK
AT+CWMODE=0-->AT+CWMODE=0
OK

sys/ztimer/core.c:159 => *** RIOT kernel panic:
                                               FAILED ASSERTION.

                                                                *** halted.
```

My Makefile has `USEMODULE += xtimer` in it.

#### Steps to reproduce the issue

I assume adding `USEMODULE += xtimer` should be sufficient because it seems like my code is not being reached before the kernel panics. One of the things things I do however is `xtimer_init();`.

#### Expected results

The code should work.

#### Actual results

The kernel panics.

#### Versions

```
Operating System Environment
----------------------------
         Operating System: Mac OS X 10.15.6
                   Kernel: Darwin 19.6.0 x86_64 i386
./dist/tools/ci/print_toolchain_versions.sh: line 78: realpath: command not found
             System shell:  (probably dash)
             make's shell: GNU bash, version 3.2.57(1)-release (x86_64-apple-darwin19)

Installed compiler toolchains
-----------------------------
               native gcc: Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
                  avr-gcc: avr-gcc (Homebrew AVR GCC 9.3.0) 9.3.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
     riscv-none-embed-gcc: riscv-none-embed-gcc (GCC) 10.1.0
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: Apple clang version 11.0.3 (clang-1103.0.32.29)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
  riscv-none-embed-newlib: "3.2.0"
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.18.2
                 cppcheck: missing
                  doxygen: 1.8.20
                      git: git version 2.28.0
                     make: GNU Make 3.81
                  openocd: Open On-Chip Debugger 0.10.0+dev-00954-gded67990 (2019-10-29-10:40)
                   python: Python 2.7.16
                  python2: Python 2.7.16
                  python3: Python 3.8.5
                   flake8: error: /usr/local/opt/python@3.8/bin/python3.8: No module named flake8
               coccinelle: missing
```

kaspar030, 2020-09-08:
So this is an assertion in ztimer, it complains about an unset clock->max_value. That smells like broken ztimer auto-config.

Could you post the output of "BOARD=hifive1b make info-modules"?

Also, it would be helpful to see how ztimer gets configured. For that, please add `#define LOG_LEVEL LOG_DEBUG` in `sys/ztimer/auto_init.c`, right before the `#include "log.h"`. Then watch the output of your application, it should print sth like this (but different) very early in the boot process:

    ztimer_init(): ZTIMER_USEC using periph timer 0, freq 1000000, width 32

mrusme, 2020-09-08:
@kaspar030 thank you for the quick reply! :) Here's the output you requested:

```
$ PATH=/usr/local/opt/binutils/bin:/usr/local/opt/make/libexec/gnubin:$PATH BOARD=hifive1b make info-modules                                                                                                                                    
auto_init
auto_init_xtimer
auto_init_ztimer
bmp388
board
cl_stack
core
core_idle_thread
core_init
core_msg
core_panic
cpu
dfp
div
fmt
frac
iolist
lora
netdev_register
netif
newlib
newlib_nano
newlib_syscalls_fe310
pca9685
periph
periph_common
periph_gpio
periph_gpio_irq
periph_i2c
periph_init
periph_init_gpio
periph_init_gpio_irq
periph_init_i2c
periph_init_plic
periph_init_pm
periph_init_rtt
periph_init_spi
periph_init_timer
periph_init_uart
periph_plic
periph_pm
periph_rtt
periph_spi
periph_timer
periph_uart
printf_float
radio
serial
si7021
si70xx
sifive_drivers_fe310
ssd1306
stdio_uart
sx1276
sx127x
sys
tmp006
tmp00x
tsl2591
xtimer
xtimer_on_ztimer
ztimer
ztimer_auto_init
ztimer_convert
ztimer_convert_frac
ztimer_convert_shift
ztimer_core
ztimer_extend
ztimer_msec
ztimer_periph_rtt
ztimer_periph_timer
ztimer_usec
```

I tried adding the `LOG_LEVEL` in `auto_init.c`:

[image]

However, the output didn't change. Kept screen attached and reset the board multiple times, still the same output as initially posted. 🤔

kaspar030, 2020-09-09:
So lora pulls in ztimer as dependency, which in turn makes xtimer use ztimer as backend. That's not bad per-se, but still somehow ztimer is not configured correctly.

> I tried adding the `LOG_LEVEL` in `auto_init.c`:
>
> [img] 
> 
> However, the output didn't change. Kept screen attached and reset the board multiple times, still the same output as initially posted. thinking

maybe ztimer_init() is not called at all. Should be called by auto_init(). you mentioned that you'd call xtimer_init(), which is usually not necessary, as auto_init would do that. is the global auto_init enabled (it should be by default)?

Can you somehow confirm that ztimer gets initialized? Maybe at a `puts("foo");` in sys/ztimer/auto_init.c `ztimer_init()`, or in the corresponding ifdef branch in sys/auto_init/auto_init.c?

kaspar030, 2020-09-09:
> auto_init
> auto_init_xtimer
> auto_init_ztimer

they're here, so unless your application messes with the regular initialization, there *should* be no issue. :) you mentioned that you're calling xtimer_init(), does that happen before main()?

mrusme, 2020-09-09:
@kaspar030 thank you for the hints here! Based on those I began digging deeper and found out that I'm loading a module (of mine) which call the following function during its initialisation:

```
void StatusDisplay::_refreshLastUpdate() {
    this->_lastUpdateSec = (uint32_t)(xtimer_now_usec() / MICROSECONDS_PER_SECOND);
    return;
}
```

If I comment out the `this->...` line, the assertion error disappears. However, when I replace that line with just `xtimer_now_usec();`, it re-appears.

Is this something that isn't supposed to be working on the `hifive1b`? Is there an alternative to that function?

kaspar030, 2020-09-09:
> Is this something that isn't supposed to be working on the `hifive1b`? Is there an alternative to that function?

it should be working. there might be an ordering issue. from where is the initialization of your module called?

xtimer cannot be used before xtimer_init(). If ztimer is used, xtimer cannot be used before ztimer_init(). Both are usually done first thing in auto_init().

mrusme, 2020-09-09:
@kaspar030 your guess was correct, it had to do with the ordering. I'm initialising the class that uses this module outside of `int main()`, so that I can use it globally across multiple functions. That seems to work fine on the nucleo boards but cause this error on the `hifive1b`.

Depending on which board is acting correctly, this issue can either be closed or rewrote into a task to make the behaviour alike, I guess?

kaspar030, 2020-09-10:
> Depending on which board is acting correctly, this issue can either be closed or rewrote into a task to make the behaviour alike, I guess?

This would maybe need documentation explaining that the auto init order has some meaning, or how to hook in application specific pre-main initialization.

Happy it works now!

closed 2020-09-10 by kaspar030