RIOT-OS/RIOT #15006: One bug may lead to NULL dereference.

tluio, 2020-09-11:
In sys/pipe/pipe_dynamic.c, if malloc returns NULL on line 41, the variable m_pipe will be NULL, but the pointer to m_pipe->pipe is returned on line 46.

tluio, 2020-09-11:
In sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c:
It can be inferred that dst can be NULL on line 1216, but it is dereferenced in gnrc_rpl_validation_DAO_ACK on line 1220.

fjmolinas, 2020-09-11:
@bergzand can you check the one targeting `usb`?

fjmolinas, 2020-09-11:
@cgundogan can you take a look at the rpl issue?

fjmolinas, 2020-09-11:
Thanks for the catch @tluio, how did you come across these bugs, did you use some kind of vulnerability scanner?

tluio, 2020-09-11:
> Thanks for the catch @tluio, how did you come across these bugs, did you use some kind of vulnerability scanner?

It's my pleasure! Yes, I used a static code analysis tool written by myself.

wosym, 2020-09-15:
@tluio  Must be quite an impressive tool, seeing that you've managed to find 3 obscure bugs in RIOT, and some more in Zephyr and some other repos in only a few days time! 
Are you planning to opensource this tool? Because I'm curious to see how it works :)

tluio, 2020-09-15:
Thanks for your interest! 
Our tool is still under development, so we consider releasing it in the future :)
Any suggestion to our bug report will be helpful to improving our tool.

Citrullin, 2020-09-28:
> Thanks for your interest!
> Our tool is still under development, so we consider releasing it in the future :)
> Any suggestion to our bug report will be helpful to improving our tool.

Sounds like a tool you may want to present on an important conference. ;) Blackhat, Chaos Communication Congress or something similar.

closed 2023-05-22 by bors[bot]