RIOT-OS/RIOT #15020: flashpage on STM32F4 CPU

biboc, 2020-09-14:
#### Description

All other CPU on STM32 board have flashpage implemented, why not on STM32F4, what's missing?

Thanks,

aabadie, 2020-09-14:
There was a tentative PR in #11681. Maybe @fjmolinas can describe what were the limitations and give some advice on how this could get in one day ?

fjmolinas, 2020-09-14:
The main issue is that `stm32f2` `stm32f4` `stm32f7` have sectors and not pages, these sectors are bigger than pages and more importantly do not have an uniform size. For one family they might be: 8kb, 8kb, 16kb, 32kb, 64kb.

My initial attempt #11681, faked pages within sectors, but it was a bit hacky, and I ended up not liking the solution at all. So currently `flashpage` needs to be adapted, or `flashpage_raw` converted to a `flash_raw`.

biboc, 2020-09-15:
Ok thanks for your answer
Yes STM32F4 works with sector, so writting a page means erasing a sector then write the page, it is difficult to standardize the process with pages
Apart from erasing problem, the function `flashpage_write` works as expected without sync problem?

fjmolinas, 2020-09-15:
> Apart from erasing problem, the function flashpage_write works as expected without sync problem?

Are you referring to the close PR? Or to current code?

biboc, 2020-09-15:
Close PR

fjmolinas, 2020-09-15:
I don't remember the exact state the PR was in when closed, I think everything was working, but erasing was hacky.

biboc, 2020-09-15:
Ok let's close this issue then

closed 2020-09-15 by biboc