RIOT-OS/RIOT #15023: gnrc: all iotlab-m3 nodes have identical short HW addresses

haukepetersen, 2020-09-15:
#### Description
Just flashed some `iotlab-m3` nodes at the Strasbourg site of the iotlab. To my surprise, all 15 of my node have an identical short HW address. The long HW addresses are unique, so no trouble there.

Example:
```
1600164691.051410;m3-2;Iface  4  HWaddr: 13:15  Channel: 26  Page: 0  NID: 0x23  PHY: O-QPSK 
1600164691.055883;m3-4;Iface  4  HWaddr: 13:15  Channel: 26  Page: 0  NID: 0x23  PHY: O-QPSK 
1600164691.060322;m3-6;Iface  4  HWaddr: 13:15  Channel: 26  Page: 0  NID: 0x23  PHY: O-QPSK 
1600164691.054618;m3-8;Iface  4  HWaddr: 13:15  Channel: 26  Page: 0  NID: 0x23  PHY: O-QPSK 
1600164691.054015;m3-20;Iface  4  HWaddr: 13:15  Channel: 26  Page: 0  NID: 0x23  PHY: O-QPSK 
1600164691.052986;m3-22;Iface  4  HWaddr: 13:15  Channel: 26  Page: 0  NID: 0x23  PHY: O-QPSK 
// and so on
```

#### Steps to reproduce the issue
flash some m3 nodes in the iotlab with `examples/gnrc_networking` and run ifconfig

#### Expected results
unique addresses

#### Actual results
non-unique addresses :-)

#### Versions
master

benpicco, 2020-09-15:
How do the long addresses look like?
I thought #14986 would have fixed this once and for all :fearful:

haukepetersen, 2020-09-15:
as said above, long addresses look just fine.

benpicco, 2020-09-15:
Yea but do they exhibit a certain pattern?

haukepetersen, 2020-09-15:
here they are (at least some of them):
```
1600164691.056362;m3-2;          Long HWaddr: 36:59:FD:65:10:6B:13:15
1600164691.056063;m3-4;          Long HWaddr: 02:54:FA:65:10:6B:13:15 
1600164691.060498;m3-6;          Long HWaddr: 02:5B:F9:65:10:6B:13:15 
1600164691.061057;m3-8;          Long HWaddr: 3A:5B:F8:65:10:6B:13:15 
```

Ups, yes they do :-) Havn't notices when not seing them next to each other...

benpicco, 2020-09-15:
Hm is that with 28d1936fae734d3f6a962ccd36ebacc05f6a4d7b ?

haukepetersen, 2020-09-15:
wait a minute, it seems #14986 is not included in my branch

haukepetersen, 2020-09-15:
reabsed my branch, now includes #14986 -> and voila, the issue is fixed. Thanks for the pointer and sorry for the fuzz!

closed 2020-09-15 by haukepetersen

benpicco, 2020-10-18:
The shuffling only brought marginally improvements - after using the iotlab testbed myself I noticed the addresses are still very similar, especially short addresses.

 #15239 should give better results.