RIOT-OS/RIOT #15028: Suspicious code in emCute

keestux, 2020-09-15:
In PR #6633, the following code was added. And the code triggers a finding by Coverity Scan.
`sys/net/application_layer/emcute/emcute.c`
```
int emcute_willupd_msg(const void *data, size_t len)
{
    assert(data && (len > 0));

    if (gateway.port == 0) {
        return EMCUTE_NOGW;
    }
    if (len > (EMCUTE_BUFSIZE - 4)) {
        return EMCUTE_OVERFLOW;
    }

    mutex_lock(&txlock);

    size_t pos = set_len(tbuf, (len + 1));
    len += (pos + 1);                                      <<<====
    tbuf[pos++] = WILLMSGUPD;
    memcpy(&tbuf[pos], data, len);

    return syncsend(WILLMSGRESP, len, true);
}
```
There is a statement `len += (pos + 1);` which I think should not be there.

If it can be explained that the statement is correct, then there is a buffer overflow if `len` is greater than `EMCUTE_BUFSIZE - 8`. And, if the statement is correct, I would like to understand why.

Citrullin, 2020-09-21:
Maybe we should add static code analysis into the CI/CD workflow to catch these kind of issues before getting them into the kernel.

keestux, 2020-09-21:
BTW. Until now @haukepetersen hasn't replied. :-(

This problem was caught by Coverity Scan. It allows setting it up via Travis or something. But I don't know how to do that.
And even we get it to do analysis (at each PR?), it still requires someone to look at the result. Not everyone feels an urge to do that.

haukepetersen, 2020-09-22:
Sorry for the delay. You are (and so is Coverity) completely right, this is a bug!

I opened a fix PR to address this: #15054

PR #15054 (2020-09-22): net/emcute: fix buffer overflow in _willupd_msg()
### Contribution description
fixes #15028

The `len` variable was indeed broken, or at least the call to `memcpy` included a buffer overflow. Should be fixed now.

### Testing procedure
- run a RIOT (e.g. `native`) node with the `examples/emcute_mqttsn` example
- run MQTT-SN gateway -> simply run `make mosquitto_rsmb` from the example application
- make sure you are using global addresses -> mosquitto_rsmb does not support link local ones...

- connect the RIOT node to tlhe gateway specifying a last will message
- update the last will message u [... cut]
closed 2020-09-22 by miri64