RIOT-OS/RIOT #15031: Cpp streams unavailable without cpp11-compat ?

nicoHarel, 2020-09-16:
Hello,

Just wanted to say: nice summit during the last 2 days , I enjoyed it =) 

Anyway, I've been working on a C/C++ project using Riot OS for some time now, and until now, I did everything without using streams. But at this point some of the code is starting to get over-complicated for no reason except the fact that I'm too lazy to really ask what's up with streams and Riot OS. 

I use an NRF52 board and the cross arm tool-chain (arm-none-eabi 7.3.1, to be exact), and every time I try to use a stringstream or any "stream object" for that matter, I get an undefined reference to __dso_handle error. Now, I know __dso_handle  is some kind of guard in cpp but why do I get this error when everything else in cpp seems to be working ? And furthermore, why using module cpp11-compat fixes the problem ? 

I'd use cpp11-compat but this makes my binary double in size...  

Thank you.

nicoHarel, 2020-09-25:
No answer ? Did I miss something obvious ? Is the large increase in the binary size considered normal (makes no sense to me) ?

kaspar030, 2020-09-25:
> I'd use cpp11-compat but this makes my binary double in size...

snarky: that's an issue of cpp - using features can be expensive... :)

I'm no c++ user at all, so I'm guessing here.
Looking at sys/cpp11-compat/cppsupport.c, I see that symbol defined there. Also, it defines some new/delete wrappers, which streams might need?

Please try to just define the symbol in your code like cppsupport.c does:
    
    void *__dso_handle __attribute__((weak)) = NULL;

maribu, 2020-09-25:
https://github.com/RIOT-OS/RIOT/blob/ad9e35c44562c2d5f7b891646508fd929ba4c487/sys/cpp11-compat/cppsupport.cpp#L29-L36

nicoHarel, 2020-09-25:
Oh I should have mentioned on my last answer that I've tried to include that symbol directly into my app code, and yes, it does work. But I get the same result (doubles the binary size). 

Well I'm guessing, based on your answers, that this is normal behaviour then. I might try to do the same thing in an other environment just to compare with RIOT, see if the size increase is similar or not.

kaspar030, 2020-09-25:
> But I get the same result (doubles the binary size).

don't forget that using streams pulls in the streams code...

maribu, 2020-09-25:
> Edit: I still nonetheless think it's weird that you have to include cpp 11 related module to make a cpp 98 feature work, don't you think?

That is true. It seems `__dso_handle` is only needed during termination of a C++ program. Maybe the C++ compiler could be persuaded to not generate any termination handling code at all? IMO, that is not useful for embedded programming anyway.

nicoHarel, 2020-09-25:
> don't forget that using streams pulls in the streams code...

Yes, you're right it seems that the increase in size is more related to the streams code in gcc-arm-none-eabi than RIOT. Now that you say it's a bit obvious that RIOT OS can't really do anything on that part. 

> IMO, that is not useful for embedded programming anyway.

Yes the streams are an advanced/fancy c++ feature and apparently, also " bigger" than I expected. Although I think using them in an embedded system makes sense depending on said system's characteristics. In my case, with 1MB of Flash, it surely isn't. And in RIOT's case, probably not either, since it's primarily targeted to be a "compact" OS.  

But a few people will still wonder why streams don't work out of the box with RIOT's c++.

maribu, 2020-09-25:
> Yes the streams are an advanced/fancy c++ feature and apparently, also " bigger" than I expected.

Sorry for the confusion. I meant that IMO termination handlers are not useful for embedded programming, as normally the firmware runs until the battery runs out or the device is power off.

I only rarely develop in C++, so I didn't want to judge on the usefulness of streams.

nicoHarel, 2020-09-25:
> Sorry for the confusion. I meant that IMO termination handlers are not useful for embedded programming, as normally the firmware runs until the battery runs out or the device is power off.
> 
> I only rarely develop in C++, so I didn't want to judge on the usefulness of streams.

Looks like a jumped the gun there a little bit =) There's kind of a heated argument in embedded c++ on when to use those kind of features or when not to.

On the matter of termination handlers, I don't know really. I wouldn't say with confidence that I'm experienced enough to discuss this. It is true that all the firmwares I have worked on always run until electrical shut-down but there might be some exceptions. So I'll leave others give their opinions.

closed 2021-07-08 by stale[bot]