RIOT-OS/RIOT #15041: bug:RIOT/cpu/stm32/include/vendor/stm32f103xe.h

kex2017, 2020-09-18:
#### Description
RIOT/cpu/stm32/include/vendor/stm32f103xe.h content is !!!stm32f101xe.h!!!

miri64, 2020-09-18:
> Do we need to backport a fix ?

On one hand-side I'd say, definitely yes. On the other, the release process for 2020.10 was already kicked off, so I am hesitant to having another point release for 2020.07, so the bug fix will just reside in the branch.

miri64, 2020-09-18:
> On one hand-side I'd say, definitely yes.

Only now looked at the change: IMHO that "fix" is too substantial to be backported, so my final answer is no ^^".

aabadie, 2020-09-18:
The change is to replace the cmsis header file by the right one. This would just be a single PR, only for 2020-07-branch because in master it's already fixed (the file no longer exists :) ).
Leaving it like this means the 2020.07 is potentially broken on stm32f103xe based boards.

miri64, 2020-09-18:
> The change is to replace the cmsis header file by the right one. This would just be a single PR, only for 2020-07-branch because in master it's already fixed (the file no longer exists :) ).
> Leaving it like this means the 2020.07 is potentially broken on stm32f103xe based boards.

Ah, so not really a backport, but a hotfix for 2020.07. That's alright, yes :-).

aabadie, 2020-09-18:
> That's alright, yes :-)

Ok, I'll open the PR. Then I leave to you to decide whether you want to publish a new point release :)

aabadie, 2020-09-18:
I made the change but I found another issue on iotlab-m3... Need to open an issue for this one (I know the culprit!).

miri64, 2020-09-18:
> I made the change but I found another issue on iotlab-m3 while testing...

Where can I find it?

aabadie, 2020-09-18:
> Where can I find it?

It's here: #15044
I was running the compile_and_test_for_board.py on iotlab-m3 and was waiting for the results (no major problems so far).

miri64, 2020-09-22:
#15044 was merged.

PR #15044 (2020-09-21): [2020.07] cpu/stm32: use correct CMSIS header for stm32f103xe line

### Contribution description

This PR replace the CMSIS header used for stm32f103xe cpu line by the valid one. In 2020.07, the CMSIS for stm32f101xe was used by mistake in #14143.

### Testing procedure

- Green Murdock
- Tests on iotlab-m3 should still work

### Issues/PRs references

fixes #15041
closed 2020-09-22 by miri64