RIOT-OS/RIOT #15060: ztimer tests fail on efm32

chrysn, 2020-09-22:
#### Description

Something seems not to work with ztimer on EFM32. I noticed this when I moved the Rust demos from xtimer over to ztimer, and both the saul_blink and the coap demos got stuck. (I have a 3000-long sleep on the ms timer, and have yet to see it return.)

Based on that, I could reproduce some error on plain RIOT, which is probably the same:

#### Steps to reproduce the issue

```shell
$ make -C tests/ztimer_overhead BOARD=stk3700 all PROGRAMMER=openocd all flash test
[...]
START
main(): This is RIOT! (Version: 2020.10-devel-1468-gc270a)
ZTIMER_USEC->adjust = 0
Timeout in expect script at "child.expect(r"min=-?\d+ max=-?\d+ avg_diff=\d+\r\n")" (tests/ztimer_overhead/tests/01-run.py:14)
```

(The board and OpenOCD combo was verified to work using the xtimer_overhead example. Same for sltb001a)

#### Expected results

On a different board, things just work:

```shell
START
main(): This is RIOT! (Version: 2020.10-devel-1468-gc270a)
ZTIMER_USEC->adjust = 0
min=6 max=6 avg_diff=6
s
```

#### Versions

Debian GNU/Linux sid, RIOT head c270aadc0

## Next steps

I'll happily look deeper into it, but would appreciate pointers at where to start.

chrysn, 2020-09-22:
Some details that may be relevant from a first look:
* The example uses modules ztimer_overhead and ztimer_usec; after dependency resolution it's
ztimer
ztimer_auto_init
ztimer_convert
ztimer_convert_frac
ztimer_convert_shift
ztimer_core
ztimer_extend
ztimer_overhead
ztimer_periph_timer
ztimer_usec
* The system doesn't lock up or anything, just the timer doesn't fire.

Next steps for me are to read utimer_nwo from a CLI tool and look around there.

kaspar030, 2020-09-22:
> I'll happily look deeper into it, but would appreciate pointers at where to start.

Looking at `boards/stk3700/include/board.h`, the stk3700 is probably using a 16bit timer at 250KHz. I skimmed the periph/timer code, it seems simple enough, and nothing jumped in my eyes that could cause issues.

First thing would be to verify how ztimer configures itself, by adding `#define LOG_LEVEL LOG_DEBUG` just before `#include "log.h"` in `sys/ztimer/auto_init.h`. 

ztimer_overhead sets a 0 delay in a loop. I`d first try if this works reliably when directly using the periph_timer API.

kaspar030, 2020-09-22:
>  utimer_nwo

whats this?

chrysn, 2020-09-22:
That was me typing too fast meaning `ztimer_now` -- but I'll hold off that approach until I've exhausted the options you've suggested.

kaspar030, 2020-09-22:
I think I got something. I found an sltb001a which uses the same periph implementation. I think ztimer sets some relative timeout using "timer_set(foo)", which ends up in periph_common's "timer_set()" implementation, ending up doing "timer_set_absolute(now+foo)". That ends up setting a value higher than the timer width.

I think the efm code needs to be fixed to write "foo & width" instead of returning -1. Let me check.

kaspar030, 2020-09-22:
This fixes it for me: https://gist.github.com/e23f2168d8b6b7c2fca1e67afe2be82a
Not sure this is a proper fix. I think xtimer is configured to use a 16bit timer even though it could to 32bit.
"corrext" would be to do sth like `value &= TIMER_TopGet(tim)`, but that would be a modulo where only `0xFFFF_FFFF` or `0xFFFF` are valid.

PR #15062 (2020-09-23): cpu/efm32/periph/timer: fix timer_set_absolute() in 16bit case

### Contribution description

Previously, the function would return -1 without setting a timer, should the absolute target time exceed the timer width (this could only happen in the 16bit case). This behaves differently than other periph_timers that use a 16bit, which implicitly sets `value & 0xFFFF`.

### Testing procedure

Test e.g., `tests/ztimer_overhead` on an affected board (sltb0001, stk3700, ...).

### Issues/PRs references

Fixes #15060.
closed 2020-09-23 by chrysn