RIOT-OS/RIOT #15080: RIOT-core is missing timer functionality

kfessel, 2020-09-24:
#### Description
while there are xtimer and ztimer in the sys module space of RIOT there is no core component that would provide a similar functionality. I consider it to be a core functionality for an IoT or Realtime capable OS to measure time an to trigger actions at certain times.

### Useful links
maybe i should check other IoT-OSes and provide some links to prove that this is core.

ya-enot, 2020-09-25:
@kfessel Consider that you don't always have hardware modules for time acquiring. What the core should do if you are not having one?

maribu, 2020-09-25:
IMO, `core` consists of:

- Platform independent code that is part of very RIOT application. This is mostly `core/sched` and `core/thread`
- code that is so tightly integrated into other `core` code, that having it elsewhere makes no sense

E.g. a network stack is also certainly a core feature of an IoT OS. But you can certainly run RIOT without any network stack if you don't need one.

So: IMO `xtimer` and `ztimer` are well placed in `sys`, rather than in `core`.

kfessel, 2020-09-25:
@ya-enot I am sure you will not find any platform i would say even MCU that you want to run RIOT on which has no timer, it might be MCU dependent but so is the switching of thread it just depend on a different part of the MCU.

@maribu network is certainly a IoT feature which is the current way RIOT is going but network is not need for a realtime embedded do something and leaveme alone MCU there for much less core than time(r), but i would consider communication a core feature (while network has do deal with different adressings schemes, comunication does not u just broadcast some message and sometimes some one responds (in UART there is typically just one listener which might respond)

x and z timer are quiet high levels and feature full timers maybe core could provide a interface which can either map to hardware if there is no timer left you can not start one or to virtual x/z timers if its the users desires.

kfessel, 2020-09-25:
I did take a short look over all RIOT-CPUs (considering that i know some of them from using them and many are cortex.. (which provides a timer in its cpu core)) they all provide some kind of timer. Most of them don't provide networking  and depend on external IC to provide networking. One could certainly say that SPI and I2C/TWI ( or RS485 which just needs some driver) are some kind of network i don't this is what @maribu had in mind).

I think this makes a clear case that timer is a core functionality (no MCU manufacturer even thought about not providing something like without a timer) at least much more than networking.

Citrullin, 2020-12-19:
@kfessel So, what is the issue here? You have to use xtimer or ztimer as a module. How does this effect you negatively?

kfessel, 2021-02-12:
@Citrullin: sorry for my late response the issue is that core modules cant depend on sys modules but to develop some features you might need timing functionality in core e.g.: switching task based on time or track their cpu-time usage.

kfessel, 2021-02-26:
once upon a time there was timer functionality in core it got removed with https://github.com/RIOT-OS/RIOT/commit/085383bfaeda67302a4359d3aecdb4555c5fb46f

maribu, 2021-02-26:
> -the solution would be a highly specific interface in core to do that manipulation (that seems like a non solution to me)

What's wrong about that? Maybe someone needs to implement round robin on a HW timer directly using [`set_timer_periodic()`](https://api.riot-os.org/group__drivers__periph__timer.html) to reduce the overhead that. The new API could easily be used by both.

closed 2022-04-18 by stale[bot]