RIOT-OS/RIOT #15102: Add whole linked list of gcoap_listener_t to coap listener

Citrullin, 2020-09-28:
#### Description
When using `gcoap_register_listener(gcoap_listener_t *listener)` with a given linked list of `gcoap_listener_t`, the function expects only one element and not a linked list. It therefore [removes the following elements](gcoap_listener_t). 
The expected behavior should be that all elements are added as coap listener.

#### Steps to reproduce the issue
1. Given a linked list `gcoap_listener_t` which contains a pointer on next to another element.
2. Use `gcoap_register_listener(gcoap_listener_t *listener)` for this linked list.

#### Expected results
The added linked list. 3 elements.

#### Actual results
2 elements, the element in `gcoap_listener->next` gets removed.

#### Versions
```
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "19.10 (Eoan Ermine)"
                   Kernel: Linux 5.3.0-22-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: "3.0.0"
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 3.7.3
                    cmake: cmake version 3.13.4
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.26.2
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01000-gdb23c13d (2020-01-10-01:06)
                   python: Python 2.7.17rc1
                  python2: Python 2.7.17rc1
                  python3: Python 3.7.5rc1
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

chrysn, 2020-10-23:
A single listener can efficiently serve multiple resources; adding several listeners is typically needed only when different software components add their listeners (in which case they wouldn't cooperate anyway), or when there's listeners with different encoders or request matchers (the former is rare, the latter [not even landed](https://github.com/RIOT-OS/RIOT/pull/14029/)).

Which use cases would benefit from that?

---

That being asked (also for the benefit of #15278 which offers to add a function to add multiple listeners), if you've been using it this way earlier you might be running into a regression I introduced in https://github.com/RIOT-OS/RIOT/pull/15147 -- the ability to add multiple registrations was missed in the checks because it's neither documented nor common (at least per the assumptions underlying the above questions).

I can follow up on the changes in #15147 (probably with little overhead; the code size savings of the PR might be negated, but at runtime it's still an enhancement), but I'd like to understand the use cases first. (Then, and depending on the regressions policy that applies here, a more informed decision on whether to fix it or to emphasize in the documentation that only one listener is added can be made)

chrysn, 2020-10-23:
Looking at #15147 in detail, even though it would have worked partially, it was certainly not intended to: While it *would* have preserved the linked list structure by appending it to the end, it would only have populated the default link_encoder for the first item, left the others at NULL, and thus would have run into the endless loops also fixed in that PR.

Given that, we're more likely not talking about a regression, and the question remains only as to the usefulness of adding a chain of listeners in one go. (If the outcome is "not useful", an assertion on the next being NULL might become useful here, for the caller may otherwise be misled to think it would do as you expected).

Citrullin, 2020-10-23:
@chrysn 
My use case was:
A user defines a gcoap resource and add the `wot_coap` module to the application. The `wot_coap` registers itself a resource where the TD is available at. Additionally to that it takes the input from the application and registers it as well. 
I already [solved it in another way](https://github.com/Citrullin/RIOT/blob/wot_plugfest_09_2020/sys/net/wot/coap/coap.c). But yes, you already defined the use-case. Having different independent modules which register their own listener. I can also imagine the following use-case as well: Having a module which generates gcoap listener from given SAUL definitions. Additionally to that a user defines endpoints for managing the devices, as well as using the `wot_coap` module. I don't see why this shouldn't work. Three independent coap endpoints. I mean the `wot_coap` needs to have all gcoap resources, in order to generate the TD from it, but that is a question about the order of the execution of this.

From my point of view it is not a big deal of not having it. It should be at least documented in a way that the user is aware of this behavior.

chrysn, 2020-10-23:
Just to understand how in the linked WoT example there could be simultaneous registrations: Where is the coap_resource ever set? If `wot_td_coap_{prop,action,event}_add` were called as a single function, can't they share an (up-to) 3 long resource array on a single listener?

In the comparison case for SAUL definitions (which I just happen to be more familiar with that WoT), I don't think that such chained listener registration would come up: If they were really registered individually, they'd be created on each registration, but the more efficient way is to have one COAP_MATCH_SUBTREE listener that extracts the parameter from the URI and then looks into whichever data structure contains all the entries, rather than keeping a resource array, a listener and the stringified form of each and every single resource around.

On documenting the behavior, I've created https://github.com/RIOT-OS/RIOT/pull/15288 -- would that work for you as a "solution" to this issue?

Citrullin, 2020-10-24:
> Just to understand how in the linked WoT example there could be simultaneous registrations: Where is the coap_resource ever set? If `wot_td_coap_{prop,action,event}_add` were called as a single function, can't they share an (up-to) 3 long resource array on a single listener?

Ohh yes, good point. It would even lead to an error with the current solutions, once there is more than one endpoint. It still has to be a linked list though, because I don't know how many resources the user adds. You can find an example of the usage [here](https://github.com/Citrullin/RIOT/blob/wot_plugfest_09_2020/tests/wot_coap/wot_coap_config.c).
It may be more efficient to do it with one array, I have to know which endpoint is a prop, action and an event. This is why this design makes more sense in my scenario. Interesting point of view though. I can probably even remove the other structs. Therefore removing the need for the user to define the content of `wot_td_prop_affordance_t`, `wot_td_int_affordance_t` and `wot_td_form_t`. Even though it removes flexibility. So, you wouldn't be able to define your own `key` for example. But from an usability standpoint this is probably even better. Removing flexibility/complexity for the user. I have to think about this. Thanks for the input.

> In the comparison case for SAUL definitions (which I just happen to be more familiar with that WoT), I don't think that such chained listener registration would come up: If they were really [...]
Can you give me an example how this could look like? For all SAUL entries just on subtree? That would be very hard to reflect for my WoT scenario, since I generate the JSON from it. That would lead to only one entry. Even though I can define the accepted structure. 

> On documenting the behavior, I've created #15288 -- would that work for you as a "solution" to this issue?
Yes, definitely better than the current behavior.

Citrullin, 2020-11-01:
Closed due to #15288

closed 2020-11-01 by Citrullin