RIOT-OS/RIOT #15123: esp32: mtd tests failing on CI

kaspar030, 2020-09-30:
#### Description

The following tests regularly fail on CI:

    tests/mtd_mapper/esp32-wroom-32:gnu
    tests/pkg_littlefs/esp32-wroom-32:gnu
    tests/pkg_littlefs2/esp32-wroom-32:gnu
    tests/pkg_spiffs/esp32-wroom-32:gnu

#### Steps to reproduce the issue

#### Expected results

#### Actual results

#### Versions

fjmolinas, 2020-09-30:
Is it reproducible locally as well?

kaspar030, 2020-09-30:
> Is it reproducible locally as well?

Does anyone have the board? Mine are in the office (where I currently never go because of pandemia).

aabadie, 2020-09-30:
> Does anyone have the board?

I have one, but won't be able to test before tomorrow.

fjmolinas, 2020-09-30:
With `tests/mtd_wrapper`:

```
2020-09-30 13:09:11,517 # rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2020-09-30 13:09:11,520 # configsip: 0, SPIWP:0xee
2020-09-30 13:09:11,526 # clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2020-09-30 13:09:11,528 # mode:DOUT, clock div:2
2020-09-30 13:09:11,530 # load:0x3fff0018,len:4
2020-09-30 13:09:11,532 # load:0x3fff001c,len:4004
2020-09-30 13:09:11,534 # load:0x40078000,len:7304
2020-09-30 13:09:11,536 # load:0x40080000,len:7224
2020-09-30 13:09:11,538 # entry 0x40080350
```

fjmolinas, 2020-09-30:
> With `tests/mtd_wrapper`:
>
> ```
> 2020-09-30 13:09:11,517 # rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> 2020-09-30 13:09:11,520 # configsip: 0, SPIWP:0xee
> 2020-09-30 13:09:11,526 # [...]

There is no issue in 2020.07, this is a regression.

fjmolinas, 2020-09-30:
I'll bisect

fjmolinas, 2020-09-30:
Git bisect is pointing to https://github.com/RIOT-OS/RIOT/pull/14754:

```
81cb769cadd31e4fb30711bf6a3da85fa88b2135 is the first bad commit
commit 81cb769cadd31e4fb30711bf6a3da85fa88b2135
Author: Leandro Lanzieri <leandro.lanzieri@haw-hamburg.de>
Date:   Mon Aug 10 14:45:39 2020 +0200

    makefiles: avoid building archives when compiling

 Makefile.base                                      | 16 ++---
 Makefile.include                                   | 75 +++++-----------------
 boards/native/Makefile.include                     |  2 -
 cpu/arm7_common/Makefile.include                   |  2 -
 cpu/cc26x0/Makefile.include                        |  2 -
 cpu/cc26x2_cc13x2/Makefile.include                 |  2 -
 cpu/efm32/families/efm32gg/Makefile.include        |  3 -
 cpu/efm32/families/efm32lg/Makefile.include        |  3 -
 cpu/efm32/families/efm32pg12b/Makefile.include     |  3 -
 cpu/efm32/families/efm32pg1b/Makefile.include      |  3 -
 cpu/efm32/families/efr32mg12p/Makefile.include     |  3 -
 cpu/efm32/families/efr32mg1p/Makefile.include      |  3 -
 cpu/esp32/Makefile.include                         | 12 ++--
 cpu/esp8266/Makefile.include                       |  4 +-
 cpu/esp_common/Makefile.include                    |  5 +-
 cpu/kinetis/Makefile.include                       |  3 -
 .../newlib_syscalls_mips_uhi/syscalls.c            |  3 +
 cpu/msp430_common/Makefile.include                 |  2 -
 cpu/native/Makefile                                |  5 +-
 makefiles/arch/atmega.inc.mk                       |  4 --
 makefiles/arch/cortexm.inc.mk                      |  7 --
 makefiles/cflags.inc.mk                            | 10 ++-
 makefiles/docker.inc.mk                            |  2 -
 makefiles/modules.inc.mk                           |  2 +-
 makefiles/scan-build.inc.mk                        |  1 -
 makefiles/vars.inc.mk                              |  3 +-
 pkg/ccn-lite/Makefile.include                      |  1 +
 pkg/cryptoauthlib/Makefile.include                 |  1 +
 pkg/jerryscript/Makefile.include                   |  3 +
 pkg/micropython/Makefile.include                   |  3 +
 pkg/nordic_softdevice_ble/Makefile.include         |  1 +
 pkg/openthread/Makefile.dep                        |  1 -
 pkg/openthread/Makefile.include                    |  3 +
 pkg/relic/Makefile.include                         |  1 +
 pkg/tlsf/Makefile.include                          |  6 --
 sys/Makefile.include                               | 11 ----
 sys/newlib_syscalls_default/Makefile.include       |  1 -
 sys/picolibc_syscalls_default/Makefile.include     |  1 -
 tests/unittests/Makefile                           |  2 +-
 39 files changed, 58 insertions(+), 157 deletions(-)
 delete mode 100644 sys/newlib_syscalls_default/Makefile.include
 delete mode 100644 sys/picolibc_syscalls_default/Makefile.include
bisect run success
```

benpicco, 2020-10-04:
Same issue as #15137

PR #15186 (2020-10-09): cpu/esp*: fix linker scripts
### Contribution description
esp32 and esp8266 specify some modules' object files on their linker scripts. #14754 changed the compilation of RIOT modules, so archives are not generated anymore. This was causing the wildcards on the linker scripts to not match the object files. This PR fixes this issue by using the folder instead of the archive name.

With these changes esp8266 is working again, and for esp32 I ran the mtd test successfully (#15123).

### Testing procedure
- both platforms should be functional again, it would be good to run the tests on the CI.
- ch [... cut]
PR #15197 (2020-10-13): cpu/esp*: fix linker scripts [backport 2020.10]
# Backport of #15186

### Contribution description
esp32 and esp8266 specify some modules' object files on their linker scripts. #14754 changed the compilation of RIOT modules, so archives are not generated anymore. This was causing the wildcards on the linker scripts to not match the object files. This PR fixes this issue by using the folder instead of the archive name.

With these changes esp8266 is working again, and for esp32 I ran the mtd test successfully (#15123).

### Testing procedure
- both platforms should be functional again, it would [... cut]
closed 2020-10-09 by benpicco