RIOT-OS/RIOT #15137: ESP8266 does not respond

Stovka, 2020-10-01:
Hello I am new to the RIOT OS. I can't get my ESP8266 working.

#### Description
After succesful flash there is no communication with ESP8266

#### Steps to reproduce the issue
main.c
```
#include <stdio.h>
#include "xtimer.h"
int main(void)
{
        puts("setup");
        while(1){
                puts("loop");
                xtimer_sleep(1);
        }
         return 0;
}
```
Makefile
```
APPLICATION = test
BOARD ?= esp8266-esp-12x
RIOTBASE ?= /home/ubuntu/RIOT
USEMODULE += xtimer
USEMODULE += esp_spiffs
include $(RIOTBASE)/Makefile.include
```
`make flash term`

#### Expected results
setup, loop, loop, ...

#### Actual results
```
ubuntu@ubutnu-VirtualBox:~/project/esp8266-1$ make flash term
Building application "test" for "esp8266-esp-12x" with MCU "esp8266".

"make" -C /home/ubuntu/RIOT/pkg/spiffs
"make" -C /home/ubuntu/RIOT/build/pkg/spiffs/src -f /home/ubuntu/RIOT/Makefile.base MODULE=spiffs
"make" -C /home/ubuntu/RIOT/boards/esp8266-esp-12x
"make" -C /home/ubuntu/RIOT/boards/common/esp8266
"make" -C /home/ubuntu/RIOT/core
"make" -C /home/ubuntu/RIOT/cpu/esp8266
"make" -C /home/ubuntu/RIOT/cpu/esp8266/freertos
"make" -C /home/ubuntu/RIOT/cpu/esp8266/periph
"make" -C /home/ubuntu/RIOT/cpu/esp8266/sdk
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/esp8266/source
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/nvs_flash/src
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/spi_flash
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/util/src
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/wpa_supplicant
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/wpa_supplicant/src/crypto
"make" -C /home/ubuntu/RIOT/cpu/esp_common
"make" -C /home/ubuntu/RIOT/cpu/esp_common/freertos
"make" -C /home/ubuntu/RIOT/cpu/esp_common/periph
"make" -C /home/ubuntu/RIOT/cpu/esp_common/vendor
"make" -C /home/ubuntu/RIOT/cpu/esp_common/vendor/xtensa
"make" -C /home/ubuntu/RIOT/drivers
"make" -C /home/ubuntu/RIOT/drivers/mtd
"make" -C /home/ubuntu/RIOT/drivers/periph_common
"make" -C /home/ubuntu/RIOT/pkg/spiffs/fs
"make" -C /home/ubuntu/RIOT/sys
"make" -C /home/ubuntu/RIOT/sys/auto_init
"make" -C /home/ubuntu/RIOT/sys/div
"make" -C /home/ubuntu/RIOT/sys/log
"make" -C /home/ubuntu/RIOT/sys/luid
"make" -C /home/ubuntu/RIOT/sys/newlib_syscalls_default
"make" -C /home/ubuntu/RIOT/sys/ps
"make" -C /home/ubuntu/RIOT/sys/random
"make" -C /home/ubuntu/RIOT/sys/random/tinymt32
"make" -C /home/ubuntu/RIOT/sys/stdio_uart
"make" -C /home/ubuntu/RIOT/sys/vfs
"make" -C /home/ubuntu/RIOT/sys/xtimer
   text	   data	    bss	    dec	    hex	filename
 275952	   5024	  29056	 310032	  4bb10	/home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/test.elf
/home/ubuntu/RIOT/dist/tools/esptool/esptool.py --chip esp8266 elf2image --flash_mode dout   --flash_size 1MB --flash_freq 26m    --version 3 -o /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/test.elf.bin /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/test.elf; printf "\n" > /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/partitions.csv; printf "nvs, data, nvs, 0x9000, 0x6000\n" >> /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/partitions.csv; printf "phy_init, data, phy, 0xf000, 0x1000\n" >> /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/partitions.csv; printf "factory, app, factory, 0x10000, " >> /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/partitions.csv; ls -l /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/test.elf.bin | awk '{ print $5 }' >> /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/partitions.csv; python3 /home/ubuntu/RIOT/dist/tools/esptool/gen_esp32part.py --verify /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/partitions.csv /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/partitions.bin
esptool.py v2.4.0
Parsing CSV input...
/home/ubuntu/RIOT/dist/tools/esptool/esptool.py --chip esp8266 --port /dev/ttyUSB0 --baud 460800 --before default_reset write_flash -z --flash_mode dout   --flash_freq 26m    0x0000 /home/ubuntu/RIOT/cpu/esp8266/bin/bootloader.bin 0x8000 /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/partitions.bin 0x10000 /home/ubuntu/project/esp8266-1/bin/esp8266-esp-12x/test.elf.bin
esptool.py v2.4.0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 50:02:91:fd:8a:1b
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0341
Compressed 9344 bytes to 6292...
Wrote 9344 bytes (6292 compressed) at 0x00000000 in 0.1 seconds (effective 503.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 85...
Wrote 3072 bytes (85 compressed) at 0x00008000 in 0.0 seconds (effective 3507.3 kbit/s)...
Hash of data verified.
Compressed 281056 bytes to 193363...
Wrote 281056 bytes (193363 compressed) at 0x00010000 in 4.4 seconds (effective 516.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/home/ubuntu/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyUSB0" -b "115200"  
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2020-10-01 10:33:36,091 # Connect to serial port /dev/ttyUSB0
Welcome to pyterm!
Type '/exit' to exit.
```

#### Versions
Operating system: Windows 10 -> VM Virtual Box -> Ubuntu 20.04.1 

I followed [this](https://api.riot-os.org/group__cpu__esp8266.html#esp8266_installation_of_esptool) guide
Everything went ok except installing coccinelle which I added from ppa. and python-serial which i used python3-serial instead
I tried both toolchain installation(riotdocker, manual) with same result
Before ESP8266 I tried arduino nano that is working fine.

```
ubuntu@ubutnu-VirtualBox:~/RIOT$ make print-versions

Operating System Environment
----------------------------
         Operating System: "Ubuntu" "20.04.1 LTS (Focal Fossa)"
                   Kernel: Linux 5.4.0-48-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0
        arm-none-eabi-gcc: missing
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: xtensa-esp8266-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: missing
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: "2.2.0"
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: missing
                 cppcheck: Cppcheck 1.90
                  doxygen: 1.8.17
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: missing
                   python: Python 2.7.18rc1
                  python2: Python 2.7.18rc1
                  python3: Python 3.8.2
                   flake8: 3.7.9 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.8.2 on Linux
               coccinelle: spatch version 1.0.8 compiled with OCaml version 4.08.1

```

wosym, 2020-10-02:
I'm not familiar with the ESP8266 at all, but does it require a separate FTDI cable? Or is it integrated in the same USB-interface you programmed it over? 
This line in your output makes me assume it does require a separate cable: 
`/home/ubuntu/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyUSB0" -b "115200"  `

Did you connect it properly?

Also: personally I would avoid doing embedded things in a VM if possible. It's always a mess to get hardware working on it.

Stovka, 2020-10-02:
It is [NODEMCU Lua ESP8266 ESP-12E CP2102](https://components101.com/development-boards/nodemcu-esp8266-pinout-features-and-datasheet#:~:text=22%20April%202020%20%2D%200%20Comments,on%20the%20ESP%2D12%20module.). It has integrated USB-interface(micro-USB connector). I also tried  ESP-01 with [this](https://makeradvisor.com/esp8266-esp-01-usb-serial-programmer/) serial programmer with same result. In VirtualBox I bridged interface: CP2102 USB to UART Bridge Controller. Also When I compile and flash from Arduino IDE(also inside VM) everything works. I think the code is not even running on chip. When I compile a flash it with /RIOT/tests/leds nothing is blinking. Flashing process looks fine, board is flashing communication LED during writing. But after reset there is nothing.

I will try to flash it from my Raspberry Pi without VM.

benpicco, 2020-10-04:
That should just work.
Can compile any of the example applications using `make BOARD=esp8266-esp-12x flash term`?

Stovka, 2020-10-04:
Unfortunately same result.
```
ubuntu@ubutnu-VirtualBox:~/RIOT/examples/hello-world$ make BOARD=esp8266-esp-12x flash term
Building application "hello-world" for "esp8266-esp-12x" with MCU "esp8266".

"make" -C /home/ubuntu/RIOT/boards/esp8266-esp-12x
"make" -C /home/ubuntu/RIOT/boards/common/esp8266
"make" -C /home/ubuntu/RIOT/core
"make" -C /home/ubuntu/RIOT/cpu/esp8266
"make" -C /home/ubuntu/RIOT/cpu/esp8266/freertos
"make" -C /home/ubuntu/RIOT/cpu/esp8266/periph
"make" -C /home/ubuntu/RIOT/cpu/esp8266/sdk
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/esp8266/source
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/nvs_flash/src
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/spi_flash
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/util/src
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/wpa_supplicant
"make" -C /home/ubuntu/RIOT/cpu/esp8266/vendor/esp-idf/wpa_supplicant/src/crypto
"make" -C /home/ubuntu/RIOT/cpu/esp_common
"make" -C /home/ubuntu/RIOT/cpu/esp_common/freertos
"make" -C /home/ubuntu/RIOT/cpu/esp_common/periph
"make" -C /home/ubuntu/RIOT/cpu/esp_common/vendor
"make" -C /home/ubuntu/RIOT/cpu/esp_common/vendor/xtensa
"make" -C /home/ubuntu/RIOT/drivers
"make" -C /home/ubuntu/RIOT/drivers/mtd
"make" -C /home/ubuntu/RIOT/drivers/periph_common
"make" -C /home/ubuntu/RIOT/sys
"make" -C /home/ubuntu/RIOT/sys/auto_init
"make" -C /home/ubuntu/RIOT/sys/log
"make" -C /home/ubuntu/RIOT/sys/luid
"make" -C /home/ubuntu/RIOT/sys/newlib_syscalls_default
"make" -C /home/ubuntu/RIOT/sys/ps
"make" -C /home/ubuntu/RIOT/sys/random
"make" -C /home/ubuntu/RIOT/sys/random/tinymt32
"make" -C /home/ubuntu/RIOT/sys/stdio_uart
   text	   data	    bss	    dec	    hex	filename
 275744	   4996	  28608	 309348	  4b864	/home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/hello-world.elf
/home/ubuntu/RIOT/dist/tools/esptool/esptool.py --chip esp8266 elf2image --flash_mode dout   --flash_size 1MB --flash_freq 26m    --version 3 -o /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/hello-world.elf.bin /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/hello-world.elf; printf "\n" > /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/partitions.csv; printf "nvs, data, nvs, 0x9000, 0x6000\n" >> /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/partitions.csv; printf "phy_init, data, phy, 0xf000, 0x1000\n" >> /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/partitions.csv; printf "factory, app, factory, 0x10000, " >> /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/partitions.csv; ls -l /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/hello-world.elf.bin | awk '{ print $5 }' >> /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/partitions.csv; python3 /home/ubuntu/RIOT/dist/tools/esptool/gen_esp32part.py --verify /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/partitions.csv /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/partitions.bin
esptool.py v2.4.0
Parsing CSV input...
/home/ubuntu/RIOT/dist/tools/esptool/esptool.py --chip esp8266 --port /dev/ttyUSB0 --baud 460800 --before default_reset write_flash -z --flash_mode dout   --flash_freq 26m    0x0000 /home/ubuntu/RIOT/cpu/esp8266/bin/bootloader.bin 0x8000 /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/partitions.bin 0x10000 /home/ubuntu/RIOT/examples/hello-world/bin/esp8266-esp-12x/hello-world.elf.bin
esptool.py v2.4.0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 50:02:91:fd:8a:1b
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0341
Compressed 9344 bytes to 6292...
Wrote 9344 bytes (6292 compressed) at 0x00000000 in 0.1 seconds (effective 509.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 84...
Wrote 3072 bytes (84 compressed) at 0x00008000 in 0.0 seconds (effective 2871.8 kbit/s)...
Hash of data verified.
Compressed 280816 bytes to 193024...
Wrote 280816 bytes (193024 compressed) at 0x00010000 in 4.3 seconds (effective 519.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/home/ubuntu/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyUSB0" -b "115200"  
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2020-10-04 17:51:54,622 # Connect to serial port /dev/ttyUSB0
Welcome to pyterm!
Type '/exit' to exit.
2020-10-04 17:52:07,098 # Serial port disconnected, waiting to get reconnected...
2020-10-04 17:52:08,100 # Serial port disconnected, waiting to get reconnected...
2020-10-04 17:52:09,102 # Serial port disconnected, waiting to get reconnected...
2020-10-04 17:52:10,103 # Serial port disconnected, waiting to get reconnected...
2020-10-04 17:52:11,105 # Try to reconnect to /dev/ttyUSB0 again...
2020-10-04 17:52:11,173 # Reconnected to serial port /dev/ttyUSB
```

benpicco, 2020-10-04:
I just tried it myself and can confirm the observation.

This is a new bug, thank you for reporting! Output works fine with `2020.07`.

I'll try a bisect.

benpicco, 2020-10-04:
> 81cb769cadd31e4fb30711bf6a3da85fa88b2135 is the first bad commit
> ```
> commit 81cb769cadd31e4fb30711bf6a3da85fa88b2135
> Author: Leandro Lanzieri <leandro.lanzieri@haw-hamburg.de>
> Date:   Mon Aug 10 14:45:39 2020 +0200
> 
>     makefiles: avoid building archives when compiling
> 
> ```

Stovka, 2020-10-04:
Perfect! I look forward to the new release.

benpicco, 2020-10-04:
Let's keep the issue open until it's fixed :wink:

PR #15186 (2020-10-09): cpu/esp*: fix linker scripts
### Contribution description
esp32 and esp8266 specify some modules' object files on their linker scripts. #14754 changed the compilation of RIOT modules, so archives are not generated anymore. This was causing the wildcards on the linker scripts to not match the object files. This PR fixes this issue by using the folder instead of the archive name.

With these changes esp8266 is working again, and for esp32 I ran the mtd test successfully (#15123).

### Testing procedure
- both platforms should be functional again, it would be good to run the tests on the CI.
- ch [... cut]
PR #15197 (2020-10-13): cpu/esp*: fix linker scripts [backport 2020.10]
# Backport of #15186

### Contribution description
esp32 and esp8266 specify some modules' object files on their linker scripts. #14754 changed the compilation of RIOT modules, so archives are not generated anymore. This was causing the wildcards on the linker scripts to not match the object files. This PR fixes this issue by using the folder instead of the archive name.

With these changes esp8266 is working again, and for esp32 I ran the mtd test successfully (#15123).

### Testing procedure
- both platforms should be functional again, it would [... cut]
closed 2020-10-09 by benpicco