RIOT-OS/RIOT #15169: Survey: Mailing lists, forum ...

miri64, 2020-10-06:
### Dear RIOTers,

  we currently evaluate new communication channels to improve discussions within the RIOT community. We consider introducing a forum, implemented in Discourse (https://www.discourse.org/).

  As your opinion matters, we kindly ask you to participate in the following brief survey, which we hope will help us to decide on the next steps:

  * https://www.unipark.de/uc/riot/bf58/

  The survey consists of max. 10 questions and should not take more than five minutes.

janosbrodbeck, 2020-10-08:
One thing to note about Gitter, since it is in the survey: Apparently element (matrix) and Gitter will merge in the long run: https://element.io/blog/gitter-is-joining-element/

waehlisch, 2020-10-24:
the survey is closed.

closed 2020-10-24 by waehlisch

kfessel, 2020-12-04:
Is there a result?

miri64, 2020-12-04:
Yes, they are posted [with the call for participation for the survey in the forum](https://forum.riot-os.org/t/survey-how-will-we-communicate-in-the-future-mailing-lists-forum/84/6) (which, if the team of maintainers has no objections will go public on Wednesday).