RIOT-OS/RIOT #15171: Failed assertion when using gnrc_minimal with SLIP

nmeum, 2020-10-07:
#### Description

I was doing some experiments with using RIOT on the HiFive1 board with a SLIP network peripheral. For this purpose, I used the `gnrc_minimal` example application and slightly modified its `Makefile` to use a static IPv6 address via SLIP (let me know if you need the exact configuration).

After some tests I ran into the following assertion:

https://github.com/RIOT-OS/RIOT/blob/f9c3e5f5a497f1214ca169a3eb0910e7bc4d19e7/sys/net/gnrc/network_layer/ipv6/nib/nib.c#L295

Here is a backtrace which I obtained using GDB:

```
#0  core_panic (crash_code=crash_code@entry=PANIC_ASSERT_FAIL, message=0x2040ea98 <assert_crash_message> "FAILED ASSERTION.") at /home/soeren/src/RIOT/core/panic.c:65
#1  0x20402020 in gnrc_ipv6_nib_handle_pkt (netif=netif@entry=0x0, ipv6=<optimized out>, icmpv6=icmpv6@entry=0x80001a90 <_pktbuf+64>, icmpv6_len=<optimized out>)
    at /home/soeren/src/RIOT/sys/net/gnrc/network_layer/ipv6/nib/nib.c:295
#2  0x2040b0f8 in gnrc_icmpv6_demux (netif=netif@entry=0x0, pkt=pkt@entry=0x80001a50 <_pktbuf>) at /home/soeren/src/RIOT/sys/net/gnrc/network_layer/icmpv6/gnrc_icmpv6.c:104
#3  0x2040164c in _demux (nh=58, pkt=0x80001a50 <_pktbuf>, netif=0x0) at /home/soeren/src/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:121
#4  _receive (pkt=0x80001a50 <_pktbuf>) at /home/soeren/src/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:901
#5  _event_loop (args=<optimized out>) at /home/soeren/src/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:201
#6  0x20400520 in sched_switch (other_prio=3) at /home/soeren/src/RIOT/core/sched.c:263
```

#### Steps to reproduce the issue

The issue is that the `netif` pointer, as passed to `_demux` function, is `NULL` and `gnrc_ipv6_nib_handle_pkt` (called later with this pointer) requires is to be non-`NULL`.

The `netif` pointer is declared in the `_receive` function of `gnrc_ipv6` as follows:

https://github.com/RIOT-OS/RIOT/blob/3b6fa61829ede55982ce057cbcf6dc205ed7cbfb/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L641

There are two functions which may assign a value to it. The first is the invocation of `gnrc_netif_hdr_get_netif`:

https://github.com/RIOT-OS/RIOT/blob/3b6fa61829ede55982ce057cbcf6dc205ed7cbfb/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L737-L740

However, the `gnrc_netif_raw` interface used by `slipdev` does not include a `GNRC_NETTYPE_NETIF` protocol type in the packet. For this reason, `gnrc_netif_hdr_get_netif` is never invoked when using SLIP. The other function which may assign a value to `netif` is `_pkt_not_for_me` as invoked from `_receive` here:

https://github.com/RIOT-OS/RIOT/blob/3b6fa61829ede55982ce057cbcf6dc205ed7cbfb/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L839

However, if `_pkt_not_for_me` returns `false` no value is assigned to `netif`. Nonetheless, `_demux` will be invoked even if `_pkt_not_for_me` returned false. If so, `netif` is NULL and `_demux` will later call `gnrc_ipv6_nib_handle_pkt` with this `NULL` pointer (see the backtrace above).

If needed I can try to assemble a sample packet, however, I believe this rewrite to be more useful to reproduce the bug.  The following change keeps the assertion from failing, but it's probably not a good fix:

```patch
diff --git a/sys/net/gnrc/network_layer/ipv6/nib/nib.c b/sys/net/gnrc/network_layer/ipv6/nib/nib.c
index bb332d70e..fb5f6201c 100644
--- a/sys/net/gnrc/network_layer/ipv6/nib/nib.c
+++ b/sys/net/gnrc/network_layer/ipv6/nib/nib.c
@@ -292,6 +292,8 @@ void gnrc_ipv6_nib_handle_pkt(gnrc_netif_t *netif, const ipv6_hdr_t *ipv6,
                               const icmpv6_hdr_t *icmpv6, size_t icmpv6_len)
 {
     DEBUG("nib: Handle packet (icmpv6->type = %u)\n", icmpv6->type);
+    if (netif == NULL)
+      return;
     assert(netif != NULL);
     gnrc_netif_acquire(netif);
     _nib_acquire();
```

#### Expected results

RIOT shouldn't crash.

#### Actual results

Failed assertion in `gnrc_ipv6_nib_handle_pkt`.

#### Versions

Given the write-up above I don't think this is needed. Using RIOT ffb7512dfca2105810ece2a13e83a0dffbbb20f3.

miri64, 2020-10-07:
> However, the `gnrc_netif_raw` interface used by `slipdev` does not include a `GNRC_NETTYPE_NETIF` protocol type in the packet.

I think this is rather the bug. If we early-exit out of the packet handler of the NIB, no neighbor discovery functionality will be working with SLIP.

nmeum, 2020-10-07:
> If we early-exit out of the packet handler of the NIB, no neighbor discovery functionality will be working with SLIP.

Yeah, I agree. Didn't intend the diff to be a proposed fix for this issue :)

miri64, 2020-10-07:
> > If we early-exit out of the packet handler of the NIB, no neighbor discovery functionality will be working with SLIP.
> 
> Yeah, I agree. Didn't intend the diff to be a proposed fix for this issue :)

Yes. Now that I read your post fully after my initial scan of your issue text I know that as well ;-).

miri64, 2020-10-07:
Could you maybe provide a fix? I'm not sure I can find the time currently to do it properly (I also don't own a hifive1, but I guess this can also be reproduced with other boards).

nmeum, 2020-10-07:
> Could you maybe provide a fix? I'm not sure I can find the time currently to do it properly

I don't think I have the time on my hands either currently. Found this issue while working on something entirely different, just wanted to document it. Maybe I will get around to it eventually but I can't promise anything.

> I guess this can also be reproduced with other boards

Yeah, if my observations are correct this can be reproduced with any board using a SLIP network interface.

miri64, 2020-10-07:
> > 
> 
> I don't think I have the time on my hands either currently. Found this issue while working on something entirely different, just wanted to document it. Maybe I will get around to it eventually but I can't promise anything.

I put it on my long TODO list then. It should be a simple fix, but I dread unforeseen consequences ^^"

nmeum, 2020-10-08:
Nice, thank you! I would also suggest changing `gnrc_ipv6` to ensure that `gnrc_netif_hdr_get_netif` is always invoked then. That is, change the `gnrc_ipv6` code to require processed packages to contain a `GNRC_NETTYPE_NETIF`:

https://github.com/RIOT-OS/RIOT/blob/3b6fa61829ede55982ce057cbcf6dc205ed7cbfb/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L737-L740

Such a change might be an additional source of unforeseen consequences though :wink:

miri64, 2020-10-09:
> Such a change might be an additional source of unforeseen consequences though wink

That's why I would make such a precondition a separate change.

miri64, 2020-10-09:
@nmeum Can you give me a setup for your application using `examples/gnrc_minimal` so I can at least try to reproduce this? Just doing this patch

```diff
diff --git a/examples/gnrc_minimal/Makefile b/examples/gnrc_minimal/Makefile
index cb657e60b4..aad38eb2cb 100644
--- a/examples/gnrc_minimal/Makefile
+++ b/examples/gnrc_minimal/Makefile
@@ -14,7 +14,7 @@ DEVELHELP ?= 0
 
 # Include packages that pull up and auto-init the link layer.
 # NOTE: 6LoWPAN will be included if IEEE802.15.4 devices are present
-USEMODULE += gnrc_netdev_default
+USEMODULE += slipdev
 USEMODULE += auto_init_gnrc_netif
 # Enable single interface optimization.
 # Remove this if more than one interface is present
```

does not seem to do it for me :-/

miri64, 2020-10-09:
Still I am not shown the address. Did you use [`GNRC_IPV6_STATIC_LLADDR`](https://doc.riot-os.org/group__net__gnrc__ipv6__conf.html#gaf9a41f698d7e8d827fe35712e3f14b64)?

miri64, 2020-10-09:
Fix in #15194.

nmeum, 2020-10-09:
> Still I am not shown the address. Did you use `GNRC_IPV6_STATIC_LLADDR`?

Yes, my configuration looks as follows:

```
IPV6_STATIC_LLADDR ?= '"fe80::cafe:cafe:cafe:1"'
CFLAGS += -DGNRC_IPV6_STATIC_LLADDR=$(IPV6_STATIC_LLADDR)

# Include SLIP package for IP over Serial communication
USEMODULE += slipdev

# Configure SLIP
CFLAGS += -DSLIPDEV_PARAM_UART=1
CFLAGS += -DSLIPDEV_PARAM_BAUDRATE=115200
```

closed 2020-10-09 by benpicco