RIOT-OS/RIOT #15172: API problem: PWM is not a low-level hardware peripheral!

hugueslarrive, 2020-10-07:
Many MCUs provide several kinds of timer with different
feature sets, for example the stm32 have basic timers,
advanced timers with up to 6 outputs taking 
complementary outpus into account, and high resolution timers with up to 12 outputs.

The pwm_conf_t data structure is based on a basic timer model
and only allows, in the case of the STM32, the implementation of only 4 outputs. It does not allow a complete description of a timer
advanced and not at all a device like the high resolution timer
with his master and his slaves.

Moreover if we configure a periph_pwm in a periph_conf.h file
we cannot configure the same timer as periph_timer so
user loses access to low-level functionality of
periph_timer and he can not  use the timers configured for periph_pwm for
something else without a custom periph_conf.h.

I think PWM shouldn't be considered a 
low-level hardware device, but as a virtual device stacked on top of a
timer device.

maribu, 2020-10-08:
I agree. There are multiple ways to generate PWM, using basic timers is only the most common implementations. (I could easily see abusing e.g. I2S and DMA to generate PWM if too few h/w timers are available. Or making use of a second CPU core on multicore platforms and bitbang PWM. Or use the SPI MOSI pin and DMA. Or ...)

Maybe the correct architecture would be a low level PWM backend API that can be implemented by any timer driver, or even as platform independent backend on top of `periph_spi` or on top of (the still to come) `periph_i2s`.

A user facing API would then multiplex the PWM channels and devices. Ideally, that should be implemented so that any overhead for device multiplexing gets optimized out when only a single backend is used.

hugueslarrive, 2020-10-08:
@maribu You read in my mind !

chrysn, 2020-10-26:
Some platforms (like nRF52) *do* have low-level PWM devices.

Could those stay with the current PWM API (and have a more high-level PWM API atop that), or would you rather see that interface converted into a driver-based one, where on some platforms a built-in driver happens to be available that isn't on others?