RIOT-OS/RIOT #15185: build: info-build doesn't work with boards without port set

bergzand, 2020-10-08:
#### Description

`make info-build` doesn't work for at least the samr21-xpro and the nucleo-f401re.

#### Steps to reproduce the issue

`BOARD=samr21-xpro make -C examples/default info-build`

#### Expected results

```
APPLICATION: default
APPDIR:      /home/koen/dev/RIOT-suit/examples/default

supported boards:
native

yada yada yada
```

#### Actual results

```
make: Entering directory '/home/koen/dev/RIOT/examples/default'
/home/koen/dev/RIOT/makefiles/info.inc.mk:95: *** No port set.  Stop.
make: Leaving directory '/home/koen/dev/RIOT/examples/default'
```

#### Versions

```
Operating System Environment
----------------------------
         Operating System: Gentoo 
                   Kernel: Linux 5.7.14-gentoo x86_64 Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz
             System shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Gentoo 9.3.0-r1 p3) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (Gentoo 9.3.0-r1 p3) 9.3.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 10.0.1 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.7.11
                    cmake: cmake version 3.16.5
                 cppcheck: Cppcheck 1.88
                  doxygen: 1.8.17
                      git: git version 2.26.2
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-ge87c637d1-dirty (2020-08-29-20:21)
                   python: Python 3.7.8
                  python2: Python 2.7.18
                  python3: Python 3.7.8
                   flake8: 3.8.1 (mccabe: 0.6.1, pycodestyle: 2.6.0, pyflakes: 2.2.0) CPython 3.7.8 on
               coccinelle: missing
```

aabadie, 2023-01-04:
Is this still an issue ? I can't reproduce it locally

```
$ make print-versions 

Operating System Environment
----------------------------
         Operating System: "Ubuntu" "22.10 (Kinetic Kudu)"
                   Kernel: Linux 5.19.0-23-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
                  avr-gcc: avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1759) 5.4.0
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: riscv-none-embed-gcc (xPack GNU RISC-V Embedded GCC, 64-bit) 10.1.0
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: Ubuntu clang version 15.0.2-1

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "4.1.0"
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: "3.2.0"
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.25.0
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.37.2
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.12.0-rc2+dev-00989-g9501b263e (2022-12-15-15:20)
                   python: Python 3.10.7
                  python2: missing
                  python3: Python 3.10.7
                   flake8: 6.0.0 (mccabe: 0.7.0, pycodestyle: 2.10.0, pyflakes: 3.0.0) CPython 3.10.7 on
               coccinelle: missing
```

maribu, 2023-05-17:
I can't reproduce either. Let's assume it was fixed.

closed 2023-05-17 by maribu