RIOT-OS/RIOT #15271: How to use lwip's socket api?

liyue75, 2020-10-22:
If I want to use lwip pkg and it's socket api.
I seem to need to port cmsis_os because lwip socket needs it, is that correct?

miri64, 2020-10-22:
Just a clarification: you don't want to use `sock`, you also don't want to use our `posix_socket` wrapper around `sock`, but you want to use the [native `socket` API of lwIP](http://www.nongnu.org/lwip/2_1_x/group__socket.html)?

miri64, 2020-10-22:
> If I want to use lwip pkg and it's socket api.
> I seem to need to port cmsis_os because lwip socket needs it, is that correct?

I don't see where this dependency would be coming from.

liyue75, 2020-10-22:
@miri64 .Yes, I want to use posix socket api. LWIP socket api has good compatibility with posix socket api. 
I want to  use lwip as a third-party module and put its in my project,
By default the mecro NO_SYS in opt.h is 0, and sys_arch.h will be included, and it will inlcude "cmsis_os.h",  this file seems like a system interface.
So should I set NO_SYS = 1?

miri64, 2020-10-22:
No, `NO_SYS` is for running lwIP bare metal (so without RIOT or any other OS, which would explain why there is a dependency to cmsis_os.h in that case). In that case you would need to go to the lwIP people, we can't help you with that.

Why can't you / don't want to use our `posix_socket` wrapper? You can have a look at the `posix_socket` example for how to use it.

If you really want to use lwIP's socket wrapper, you need to enable `LWIP_SOCKET=1` in `pkg/lwip/include/lwipopts.h` and add `-Ipkg/lwip/src/include/compat/posix` to `INCLUDES` in `pkg/lwip/Makefile.include`. Be aware however, that this is entirely untested.

liyue75, 2020-10-22:
@miri64 Thanks a lot.
The reason why I can't use your posix_socket is that: I want to port another program, Micro-XRCE-DDS-Client
[https://github.com/eProsima/Micro-XRCE-DDS-Client](url), to RIOT, and It uses posix-like socket interfaces. 
I know that lwip has a good compatibility for the socket.
But I don't know if it works well using RIOT's posix_socket wrapper, for example, I can not find function "getaddrinfo" in RIOT for stm32 boards.
 I will try using RIOT's posix_socket wrapper, and maybe make some changes, because it's a lot easier than 
porting cmsis_os.

miri64, 2020-10-22:
> But I don't know if it works well using RIOT's posix_socket wrapper, for example, I can not find function "getaddrinfo" in RIOT for stm32 boards.

As we have a `sock_dns` client, porting `getaddrinfo` should be a simple addition.

miri64, 2020-10-22:
(but to make the discussion short: yes, not all functions defined in POSIX are there yet, but the functionality to implement them, is there)

liyue75, 2020-10-22:
@miri64 Thanks for your reply,
 now I will close this comment.

closed 2020-10-22 by liyue75