RIOT-OS/RIOT #15276: fatal error: bits/libc-header-start.h: No such file or directory

nuaaysh, 2020-10-22:
Hi,
    I meet one question when trying RIOT on Ubuntu:
``
shyuan@shyuan:\~/GitHub/RIOT$ git checkout <LATEST_RELEASE>
bash: syntax error near unexpected token `newline'
shyuan@shyuan:\~/GitHub/RIOT$ sudo ./dist/tools/tapsetup/tapsetup
[sudo] password for shyuan: 
creating bridge tapbr0
creating tap0
creating tap1
shyuan@shyuan:\~/GitHub/RIOT$ cd examples/default/
shyuan@shyuan:\~/GitHub/RIOT/examples/default$ make all
Building application "default" for "native" with MCU "native".

In file included from /home/shyuan/GitHub/RIOT/examples/default/main.c:25:
/usr/include/stdio.h:27:10: fatal error: bits/libc-header-start.h: No such file or directory
   27 | #include <bits/libc-header-start.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [/home/shyuan/GitHub/RIOT/Makefile.base:107: /home/shyuan/GitHub/RIOT/examples/default/bin/native/application_default/main.o] Error 1
make: *** [/home/shyuan/GitHub/RIOT/examples/default/../../Makefile.include:603: application_default.module] Error 2
``
I print other information:
``
shyuan@shyuan:~/GitHub/RIOT$ make print-versions

Operating System Environment
         Operating System: "Ubuntu" "20.04.1 LTS (Focal Fossa)"
                   Kernel: Linux 5.4.0-52-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
               native gcc: gcc (GCC) 9.3.1 20200430 (for GNAT Community 2020 20200429)
        arm-none-eabi-gcc: missing
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
     arm-none-eabi-newlib: missing
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
                   ccache: missing
                    cmake: cmake version 3.16.3
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: missing
                   python: Python 2.7.18rc1
                  python2: Python 2.7.18rc1
                  python3: Python 3.8.5
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
``

pokgak, 2020-10-22:
Hi! Have you installed `gcc-multilib` using apt?

nuaaysh, 2020-10-22:
> Hi! Have you installed `gcc-multilib` using apt?

Thanks for your suggestions.
I try it, now I get a new error:
`
shyuan@shyuan:~/GitHub/RIOT/examples/default$ make all
Building application "default" for "native" with MCU "native".

"make" -C /home/shyuan/GitHub/RIOT/boards/native
"make" -C /home/shyuan/GitHub/RIOT/boards/native/drivers\
...
"make" -C /home/shyuan/GitHub/RIOT/sys/xtimer
/home/shyuan/opt/GNAT/2020/bin/../libexec/gcc/x86_64-pc-linux-gnu/9.3.1/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libm.so when searching for -lm
/home/shyuan/opt/GNAT/2020/bin/../libexec/gcc/x86_64-pc-linux-gnu/9.3.1/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libm.a when searching for -lm
/home/shyuan/opt/GNAT/2020/bin/../libexec/gcc/x86_64-pc-linux-gnu/9.3.1/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libdl.so when searching for -ldl
/home/shyuan/opt/GNAT/2020/bin/../libexec/gcc/x86_64-pc-linux-gnu/9.3.1/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libdl.a when searching for -ldl
/home/shyuan/opt/GNAT/2020/bin/../libexec/gcc/x86_64-pc-linux-gnu/9.3.1/ld: skipping incompatible /home/shyuan/opt/GNAT/2020/bin/../lib/gcc/x86_64-pc-linux-gnu/9.3.1/libgcc.a when searching for -lgcc
/home/shyuan/opt/GNAT/2020/bin/../libexec/gcc/x86_64-pc-linux-gnu/9.3.1/ld: cannot find -lgcc
/home/shyuan/opt/GNAT/2020/bin/../libexec/gcc/x86_64-pc-linux-gnu/9.3.1/ld: cannot find -lgcc_s
/home/shyuan/opt/GNAT/2020/bin/../libexec/gcc/x86_64-pc-linux-gnu/9.3.1/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libc.so when searching for -lc
/home/shyuan/opt/GNAT/2020/bin/../libexec/gcc/x86_64-pc-linux-gnu/9.3.1/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libc.a when searching for -lc
collect2: error: ld returned 1 exit status
make: *** [/home/shyuan/GitHub/RIOT/examples/default/../../Makefile.include:594: /home/shyuan/GitHub/RIOT/examples/default/bin/native/default.elf] Error 1
`
Could you please give me suggestions?

Thanks a lot!

Best wishes,

jeandudey, 2021-03-20:
@nuaaysh Looks like there's another compiler in the path that it's interferring with the resolution of the compiler needed.

Specifically this path contains the compiler: 
- `/home/shyuan/opt/GNAT/2020`

Could you confirm it still happens to you?

nuaaysh, 2021-03-26:
@jeandudey 
Thanks a lot! Because I installed Ada before, there are two gcc version:default gcc and ada gcc. 

After modifying the path, everything is OK!

Best wishes,

closed 2021-03-26 by nuaaysh