RIOT-OS/RIOT #15309: Packages outside RIOT tree

jdavid, 2020-10-27:
#### Description
It's not possible to make packages outside of RIOT's tree, the way it can be done for modules with `EXTERNAL_MODULE_DIRS`. I get:

    make[1]: *** /home/jdavid/sandboxes/UiO/wsn_riot/RIOT/pkg/arduino_sparkfun_bme280: No such file or directory.  Stop.

I added the package to our own tree `wsn_riot/pkg` but it looks for the package in RIOT's tree. Tried adding `EXTERNAL_MODULE_DIRS`, didn't find a way in the documentation.

We have to package many Arduino libraries for sensors and it would be easier to do if it was possible to package them out of RIOT's tree, as it's possible to do with sys modules and drivers. This would help others that need the same.

### Useful links

None. This concerns the RIOT build system.

NikLeberg, 2021-11-15:
Hi there. I'm currently getting up to speed with RIOT and also ran into this issue. I'd like to provide external pkg dirs similar to external modules.
Ultimately I'd love to contribute my created packages and modules back to RIOT with a pr. But while developing them it would be convenient if there was a way to add them from outside of the riot tree.

@jdavid did you find a workaround?

jdavid, 2021-11-15:
@NikLeberg No. Probably the easiest is to keep a fork of RIOT.

NikLeberg, 2021-11-15:
I've hacked together a (very bad) patch that allows setting EXTERNAL_PKG_DIRS. See: [0001-Makefile-add-EXTERNAL_PKG_DIRS.zip](https://github.com/RIOT-OS/RIOT/files/7542034/0001-Makefile-add-EXTERNAL_PKG_DIRS.zip)

Maybe it's of interest.

closed 2022-02-07 by fjmolinas