RIOT-OS/RIOT #15311: Arduino's Serial.print(float) doesn't work

jdavid, 2020-10-27:
#### Description

Arduino's `Serial.print(float)` doesn't print anything.

#### Steps to reproduce the issue

For example I modified `tests/sys_arduino/arduino-test.sketch` to test floats as well, this is the diff:

```
iff --git tests/sys_arduino/arduino-test.sketch tests/sys_arduino/arduino-test.sketch
index 1aa30a9cb..59c2da2c9 100644
--- tests/sys_arduino/arduino-test.sketch
+++ tests/sys_arduino/arduino-test.sketch
@@ -105,6 +105,10 @@ static void print_test(void)
         Serial.print("): ");
         Serial.println(ul, f);
     }
+
+    Serial.print("print(float): ");
+    Serial.print((float)3.14);
+    Serial.println();
 }
 
 void loop(void)
```

#### Expected results

```
2020-10-27 12:03:55,271 # Hello Arduino!
print
2020-10-27 12:03:57,307 # print(int, BIN): 1111101011000111
[...]
2020-10-27 12:03:58,441 # println(unsigned long, HEX): 499602d2
2020-10-27 12:03:58,458 # print(float): 3.14
```

#### Actual results

This is the output I get, tested with Arduino Uno:

```
2020-10-27 12:03:55,271 # Hello Arduino!
print
2020-10-27 12:03:57,307 # print(int, BIN): 1111101011000111
[...]
2020-10-27 12:03:58,441 # println(unsigned long, HEX): 499602d2
2020-10-27 12:03:58,458 # print(float): 
```

#### Versions

```
Operating System Environment
----------------------------
         Operating System: Gentoo 
                   Kernel: Linux 5.4.66-gentoo x86_64 Intel(R) Core(TM) i3-5010U CPU @ 2.10GHz
             System shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Gentoo 9.3.0-r1 p3) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
                  avr-gcc: avr-gcc (Gentoo 9.3.0-r1 p3) 9.3.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 10.0.1 

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.17.4
                 cppcheck: Cppcheck 1.88
                  doxygen: 1.8.17
                      git: git version 2.26.2
                     make: GNU Make 4.2.1
                  openocd: missing
                   python: Python 3.7.9
                  python2: Python 2.7.18
                  python3: Python 3.7.9
                   flake8: 3.8.1 (mccabe: 0.6.1, pycodestyle: 2.6.0, pyflakes: 2.2.0) CPython 3.7.9 on
               coccinelle: missing
```

aabadie, 2020-10-27:
#15314 is fixing this issue, can you confirm @jdavid ?

jdavid, 2020-10-28:
@aabadie The test doesn't fit into memory anymore (Arduino Uno), so I had to comment the lines above. But yes, it works!

aabadie, 2020-10-28:
> The test doesn't fit into memory anymore (Arduino Uno)

Strange, it fits when I build in Docker (as reported in #15314). What is the version of your AVR toolchain ?

jdavid, 2020-10-28:
```
$ avr-g++ --version
avr-g++ (Gentoo 9.3.0-r1 p3) 9.3.0
```

```
$ eix-installed -a | grep avr
cross-avr/avr-libc-2.0.0
cross-avr/binutils-2.34-r2
cross-avr/gcc-9.3.0-r1
dev-embedded/avrdude-6.3
```

aabadie, 2020-10-28:
Ok, so your version is much more recent than the default RIOT one:
```
$ docker run --rm -ti riot/riotbuild avr-g++ --version
avr-g++ (GCC) 5.4.0
[...]
```

This explains the difference in code size I think.

PR #15314 (2020-11-17): sys/arduino: fix print float

### Contribution description

This PR is an attempt to fix #15311 

As reported on #15311 I confirmed that `Serial::print(float)` doesn't work on Arduino but also on ARM (tested on nucleo-l073rz). The proposed fix in this PR is to use `fmt_float` instead of `sprintf`.
The size on ROM is increased significantly:

I'm using the following patch on master:

```diff
diff --git a/tests/sys_arduino/arduino-test.sketch b/tests/sys_arduino/arduino-test.sketch
index 1aa30a9cb4..a782af680a 100644
--- a/tests/sys_arduino/arduino-test.sketch
+++ b/tests/sys_arduino/arduino-tes [... cut]
closed 2020-11-17 by maribu