RIOT-OS/RIOT #15313: [RFC] Towards an IEEE 802.15.4 MAC layer

jia200x, 2020-10-27:
### Description

I open this issue to specify a roadmap for a standard IEEE 802.15.4 MAC layer implementation.

This is desirable for several reasons:
- The IEEE 802.15.4 MAC is designed for low power devices. This would indeed improve our "radios always listening" situation, since the standard describes several mechanisms for receiving data "only when it's needed".
- Security is part of the standard (authentication, encryption).
- Interoperability with other IEEE 802.15.4 devices (e.g [Full MAC devices](https://www.cascoda.com/products/chips/ca-8210/))
- It can be used to implement other standards on top (e.g 6TiSCH, which runs on top of the TSCH mode of IEEE 802.15.4)
- Device discovery (scanning)
- Support for Full-function devices (FFD) and Reduced-function devices (RFD).

### Required steps

Since the standard is retro-compatible, we could focus on implementing "some" parts of the IEEE 802.15.4 2006 standard and keep extending as needed. At some point we should be able to introduce primitives of the 2015 standard and amendments (e.g 2012e, TSCH).

The MAC layer can be implemented using the IEEE 802.15.4 SAP API (see #11324 ). The minimal IEEE 802.15.4 MAC layer would only require to implement the "non-beacon" operation, since it's the one used for 6LowPAN. Besides that, we would need:
- Coordinator roles (PAN coordinator, etc)
- PAN association (joining a coordinator)
- Receive / send data inside a PAN (e.g use short address when possible, etc)
- Scanning (only active mode is required, since it would operate as a beacon-less network), to detect coordinators.

In the mid/long term, some desirable features:
- Indirect Transmission (a coordinator maintains a queue of frames for RFD nodes, and the RFD node polls periodically to check if there are frames for it).
- Security (authentication, encryption) (it already started in #15150 )
- Beacon-enabled operations ("superframes", Guaranteed Time Slots, etc).
- TSCH mode :)

#### Architecture

Note the MAC layer is independent of the network stack:
```
+-------------------------+        +-------------------------+
|                         |        |                         |
|        GNRC Netif       |        |        LWIP             |
|                         |        |                         |
+-------------------------+        +-------------------------+
           ^                                   ^                                        
           |MCPS/MLME                          |MCPS/MLME                                        
           |                                   |                                        
           v                                   v                                        
+------------------------------------------------------------+
|                                                            |
|                        IEEE 802.15.4 MAC                   |
|                                                            |
+------------------------------------------------------------+
                               ^                             
                               | SubMAC API                            
                               v                             
+------------------------------------------------------------+
|                                                            |
|                        IEEE 802.15.4 SubMAC                |
|                                                            |
+------------------------------------------------------------+
                               ^                             
                               | Radio Ops/                            
                               v Event Notification                            
+------------------------------------------------------------+
|                                                            |
|                        IEEE 802.15.4 Radio HAL             |
|                                                            |
+------------------------------------------------------------+
```

### Roadmap

- [ ] Implement MCPS-DATA primitive (send/receive data)- 
- [ ] Implement PAN coordinator and coordinator role
- [ ] Implement association mechanism
- [ ] Implement Scanning
- TBD

jia200x, 2020-10-27:
I think @maribu, @benpicco, @PeterKietzmann , @miri64, @fjmolinas and @kaspar030 might be interested in this

jia200x, 2020-10-27:
and probably @chrysn

miri64, 2022-04-18:
@jia200x how is the roadmap going ahead for this one?

jia200x, 2022-04-19:
kind of ongoing, but with a new direction.
I think it doesn't make sense anymore to implement a MAC from scratch, but rather re-use existing MAC implementations (e.g openDSME) and continue development from there.

miri64, 2022-04-19:
Then can this be closed?

jia200x, 2022-04-19:
let's leave it open. It just needs to be updated, but the roadmap should still be here