RIOT-OS/RIOT #15330: ztimer: timer not removed on arduino-due

pokgak, 2020-10-28:
#### Description

I encountered a weird bug when using ztimer on arduino-due but I cannot reproduce the bug on other boards. I have a shell cmd that sets a timer and directly removed it after. When the shell command is called for the first time, it works as expected and the callback is not triggered but when I called the shell command again, the callback will trigger immediately even though it's the same command. A test application to reproduce this can be found [here](https://github.com/pokgak/RIOT/tree/ztimer-remove-arduino-due/examples/test_ztimer).

#### Steps to reproduce the issue

1. Get the test application
```
git clone https://github.com/pokgak/RIOT -b ztimer-remove-arduino-due
```

2. Build and flash for arduino-due
```
BOARD=arduino-due make -C examples/test_ztimer flash term
```
3. Run the `test` shell cmd twice.

#### Expected results
Nothing happens, the set timer is removed.

#### Actual results
The callback triggered. The test application will fail the exception.

#### Versions

RIOT master (912d82dc40d0720dee3523a0d1d7078f24926704)

```
pi@HIL-R3P8:~/.tmp/RobotFW-tests/RIOT $ make print-versions

Operating System Environment
----------------------------
         Operating System: "Raspbian GNU/Linux" "10 (buster)"
                   Kernel: Linux 4.19.118-v7+ armv7l unknown
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.13.4
                 cppcheck: Cppcheck 1.86
                  doxygen: 1.8.13
                      git: git version 2.20.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-00925-g08195413 (2019-08-29-11:27)
                   python: Python 2.7.16
                  python2: Python 2.7.16
                  python3: Python 3.7.3
                   flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.7.3 on Linux
               coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

kaspar030, 2020-10-28:
Thanks for the working reproducer.
I found the issue: the sam3x periph_timer `timer_clear()` disables the timer's interrupt, but does not clear the corresponding flag. So when a subsequent `timer_set_absolute()` re-enables the interrupt, the former compare would trigger instantly.
If you run the two "test" commands of the test application very fast (within the 1000000us of the initial timer set), it doesn't crash.

This workaround fixes it for me: https://gist.github.com/kaspar030/16ee3a700780573e2cab6305d9842132

The timer event flag register is clear-on-read, the workaround reads the status register to clear previous flags.

But there's an issue with the work around: the register contains the flags for *all* channels, thus any timer_set() on one channel might clear the flag of another... I think. We might want to limit this to only one channel (for now), to be safe.

Does anybody feel like dealing with this?

pokgak, 2020-10-29:
If I understand correctly, the gist you suggested does not touch any timer specific code. I tested the same application using xtimer instead of ztimer, converting the ztimer functions to its analogous functions in xtimer and the same problem does not occur. I expect the same problem to also affect xtimer. Does this mean xtimer already "fixed" this bug somewhere in its implementation?

kaspar030, 2020-10-29:
> Does this mean xtimer already "fixed" this bug somewhere in its implementation?

no, xtimer always has a low level timer running, so it doesn't trigger the issue:

https://github.com/RIOT-OS/RIOT/blob/842f0c2a6b2623e185ac41bc66b17d33b9c61353/sys/xtimer/xtimer_core.c#L157-L160

ztimer doesn't need to detect cycles for 32bit low level timers.

pokgak, 2020-10-29:
I see. Thanks for the explanation!

kaspar030, 2020-10-29:
> We might want to limit this to only one channel (for now), to be safe.

This can be done by setting `TIMER_CHANNEL_NUMOF` to 1.

kaspar030, 2020-10-30:
> Does anybody feel like dealing with this?

Fix in https://github.com/RIOT-OS/RIOT/pull/15349.

closed 2020-10-30 by bergzand