RIOT-OS/RIOT #15353: uhcp: Incorrect handling of uhcp_push_t prefix

nmeum, 2020-10-30:
#### Description

The `uhcp_handle_push` function from the `uhcp` module contains a bug in the handling of the prefix contained in the `uhcp_push_t` type. This bug causes a stack-based buffer overflow. The bug is in the following lines:

https://github.com/RIOT-OS/RIOT/blob/f163f1580b7ef86cdd9d7dfab7f9a7ef8d17ec71/sys/net/application_layer/uhcp/uhcp.c#L101-L103

Consider a `prefix_bytes` value with `prefix_bytes < 16`. For example `prefix_bytes == 0`, with this specific value `memset(3)` would be invoked as:

```C
memset(prefix + 16, '\0', 16);
```

Since `sizeof(prefix) == 16` this would write `16` null bytes on the stack after the `prefix` buffer:

https://github.com/RIOT-OS/RIOT/blob/f163f1580b7ef86cdd9d7dfab7f9a7ef8d17ec71/sys/net/application_layer/uhcp/uhcp.c#L100

#### Steps to reproduce the issue

Use the following example application:

```C
#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>
#include <panic.h>

#include <netinet/in.h>
#include <net/uhcp.h>

static uint8_t buf[] = {
    85, 72, 67, 80, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0,
};

void uhcp_handle_prefix(uint8_t *prefix, uint8_t prefix_len, uint16_t lifetime,
                        uint8_t *src, uhcp_iface_t iface) {
    (void)prefix;
    (void)prefix_len;
    (void)lifetime;
    (void)src;
    (void)iface;
    return;
}

int main(void)
{
    uint8_t addr_buf[16];

    memset(addr_buf, '\0', 16);
    uhcp_handle_udp(buf, sizeof(buf), &addr_buf[0], 23, 0);

    exit(EXIT_SUCCESS);
    return 0;
}
```

With the following `Makefile`:

```
APPLICATION = uhcp
BOARD = native

# Required modules
USEMODULE += gnrc_ipv6
USEMODULE += gnrc_sock_udp
USEMODULE += uhcpc

# Enable DHCP client
CFLAGS += -DUHCP_CLIENT

RIOTBASE ?= $(CURDIR)/../..
include $(RIOTBASE)/Makefile.include
```

Compile with:

```
$ make BOARD=native -C examples/uhcpc/ all-asan
```

Run with:

```
$ make BOARD=native -C examples/uhcpc/ term
```

#### Expected results

The application should not crash.

#### Actual results

The application crashes with the following ASAN error:

```
==31237==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x565d2780 at pc 0xf7a25c26 bp 0x565d2718 sp 0x565d22f0
WRITE of size 16 at 0x565d2780 thread T0
    #0 0xf7a25c25  (/lib32/libasan.so.5+0xa9c25)
    #1 0x565b7bb5 in uhcp_handle_push /home/nmeum/RIOT/sys/net/application_layer/uhcp/uhcp.c:102
    #2 0x565b79d9 in uhcp_handle_udp /home/nmeum/RIOT/sys/net/application_layer/uhcp/uhcp.c:63
    #3 0x5658dc24 in main /home/nmeum/RIOT/examples/uhcpc/main.c:33
    #4 0x5658df91 in main_trampoline /home/nmeum/RIOT/core/init.c:57
    #5 0xf76d353a in makecontext (/lib/i386-linux-gnu/libc.so.6+0x4153a)

0x565d2780 is located 11200 bytes inside of global variable 'main_stack' defined in '/home/nmeum/RIOT/core/init.c:62:13' (0x565cfbc0) of size 12288
SUMMARY: AddressSanitizer: stack-buffer-overflow (/lib32/libasan.so.5+0xa9c25) 
Shadow bytes around the buggy address:
  0x2acba4a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x2acba4b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x2acba4c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x2acba4d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x2acba4e0: 00 00 00 00 00 00 00 00 00 00 f1 f1 f1 f1 00 00
=>0x2acba4f0:[f2]f2 f2 f2 f2 f2 00 00 00 00 00 06 f2 f2 f2 f2
  0x2acba500: f2 f2 00 00 00 00 00 06 f2 f2 f3 f3 f3 f3 00 00
  0x2acba510: 00 00 00 00 00 00 00 00 f1 f1 f1 f1 00 00 00 00
  0x2acba520: 00 06 f2 f2 f3 f3 f3 f3 00 00 00 00 00 00 00 00
  0x2acba530: f1 f1 f1 f1 00 00 f2 f2 f3 f3 f3 f3 00 00 00 00
  0x2acba540: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==31237==ABORTING
```

This exactly the case where `prefix_bytes == 0` laid out above.

#### Versions

I don't think this is needed to reproduce the issue, can provide if needed.

miri64, 2020-10-30:
Ah no.... only just understood that `prefix_len == 0` was just an example... I think the simplest fix might just be

```diff
diff --git a/sys/net/application_layer/uhcp/uhcp.c b/sys/net/application_layer/uhcp/uhcp.c
index 3c5b554b49..507ce84671 100644
--- a/sys/net/application_layer/uhcp/uhcp.c
+++ b/sys/net/application_layer/uhcp/uhcp.c
@@ -97,9 +97,11 @@ void uhcp_handle_push(uhcp_push_t *req, uint8_t *src, uint16_t port, uhcp_iface_
     char addr_str[INET6_ADDRSTRLEN];
     char prefix_str[INET6_ADDRSTRLEN];
     inet_ntop(AF_INET6, src, addr_str, INET6_ADDRSTRLEN);
-    uint8_t prefix[16];
+    uint8_t prefix[IN6ADDRSZ] = { 0 };
+    if (req->prefix_len == 0) {
+        return;
+    }
     size_t prefix_bytes = (req->prefix_len + 7)>>3;
-    memset(prefix + 16 - prefix_bytes, '\0', 16 - prefix_bytes);
     memcpy(prefix, req->prefix, prefix_bytes);
 
     inet_ntop(AF_INET6, prefix, prefix_str, INET6_ADDRSTRLEN);
```

nmeum, 2020-10-31:
To be honest: I personally don't understand this code in its entirety, I have a brief understanding of what the code is supposed to do but the specific implementation details are beyond me (adding some comments to the code would IMHO be wise). Otherwise, I would have come up with a patch myself. For this reason, I can't really comment on how to fix this properly. The code in question seems to have been added 5 years ago by @kaspar030 in  4fb07a852dd32fb578850ab672de899e46536267, maybe he has an idea how to fix it?

miri64, 2020-11-02:
As the idea is to have a prefix, initializing the whole array with 0 first, instead of setting only the bytes that are not overwritten later, seems to me the simplest solution. Not really sure, why it was made in this chopped up way before, except maybe timing, which isn't really crucial here.

closed 2020-11-24 by benpicco