RIOT-OS/RIOT #15356: Does riot has parallel interface driver?

liyue75, 2020-10-31:
For boards like stm32f4, they have FSMC(flexible static memory controller) which is used to interface with memories through parallel port. Does riot has driver for FSMC or parallel interface?

benpicco, 2020-11-14:
Is this comparable with QSPI?

liyue75, 2020-11-16:
@benpicco Maybe I didn' t describe it clearly. Which I mean  is parallel bus which use address lines and data lines, STM32 has a peripheral to control these lines called FSMC.
Now I can use the FSMC to control the parallel data transmission. It seems there is no FSMC driver in RIOT.

benpicco, 2020-11-16:
No unfortunately there is no driver yet, you would need to provide one yourself.
You can compare what they are doing over at Zephyr: https://github.com/zephyrproject-rtos/zephyr/pull/29686

liyue75, 2020-11-16:
@benpicco Thank you for your reply, I have alreadly used it.
Now I will close this issue.

closed 2020-11-16 by liyue75