RIOT-OS/RIOT #15357: Link failures since 81cb769

larseggert, 2020-11-02:
#### Description

81cb769cadd31e4fb30711bf6a3da85fa88b2135 broke the way I build [quant for RIOT](https://github.com/NTAP/quant/tree/master/riot) (confirmed via `git bissect`):

The link step doesn't seem to include the sources added to `SRC`, causing the link failure below. (The files do get compiled, just not linked.)

#### Steps to reproduce the issue

1. `git clone --shallow git@github.com:NTAP/quant.git`
2. `git submodule update --init --recursive --depth 1`
3. `cd riot`
4. set `RIOTBASE` correctly
5. `make BUILD_IN_DOCKER=1`

#### Expected results

Expected the firmware to build and link.

#### Actual results

```
/data/riotbuild/riotproject/riot/bin/esp32-wroom-32/application_quant-client/main.o:(.text.startup.main+0x4): undefined reference to `quic_transaction'
/data/riotbuild/riotproject/riot/bin/esp32-wroom-32/application_quant-client/main.o: In function `main':
/data/riotbuild/riotproject/riot/main.c:40: undefined reference to `quic_transaction'
collect2: error: ld returned 1 exit status
/data/riotbuild/riotbase/Makefile.include:584: recipe for target '/data/riotbuild/riotproject/riot/bin/esp32-wroom-32/quant-client.elf' failed
make: *** [/data/riotbuild/riotproject/riot/bin/esp32-wroom-32/quant-client.elf] Error 1
make: *** [/Users/lars/Documents/Code/riot/makefiles/docker.inc.mk:304: ..in-docker-container] Error 2
```

leandrolanzieri, 2020-11-03:
@larseggert thanks for reporting this. Could you give it a try with #15364?

larseggert, 2020-11-03:
#15364 fixes the issue!

PR #15364 (2020-11-04): Makefile.include: discover object files in sub-paths while linking
### Contribution description
The `find` command that collects the object files while linking is currently too strict and does not include sub-directories within the module's directories. This causes use cases like the described in #15357 to fail. This allows a recursive search for object files in those directories.

I've tested this following the steps in #15357 and now it works properly.

### Testing procedure
- Green murdock
- The issue described in #15357 should be fixed with this.

### Issues/PRs references
Fixes #15357
Iss [... cut]
closed 2020-11-04 by cgundogan