RIOT-OS/RIOT #15400: Feature Request:cpu load/utlization

ad6sh, 2020-11-07:
#### Description
I am implementing an application using RIOT where I need to get features about a particular device such as board,runtime and network interfaces information.I need to get the cpu load of the device at present using a command,just like in UNIX like systems,for eg. using "w" or "top" command. Is there anyway to get such metrics in RIOT ?

### Useful links
https://en.wikipedia.org/wiki/Load_(computing)

benpicco, 2020-11-14:
Does the [`schedstatistics`](https://doc.riot-os.org/group__schedstatistics.html) module fulfill your requirements?

ad6sh, 2021-01-20:
Almost! I could calculate the CPU utilization using runtime ticks of each thread.CPU load could not be calculated like in Unix like systems,Where exponentially weighted moving averaging  of  sched_num_threads is required. :)

https://en.wikipedia.org/wiki/Load_(computing)

closed 2022-04-18 by stale[bot]