RIOT-OS/RIOT #15434: stm32: cmsis: spurious build issues

cgundogan, 2020-11-12:
#### Description
I notice spurious build issues when building an stm32 based board on the CI logs for quite some time now.  The issue is with the cmsis vendor headers, which are cloned during the build process. 

An example log is here: https://ci.riot-os.org/RIOT-OS/RIOT/15337/bf9f5cce1f0c77cab947bb3c37d35516c5ed3733/output/compile/tests/riotboot/nucleo-l552ze-q:gnu.txt

#### Steps to reproduce the issue
I can also reproduce this problem locally on my machine with the following command:
```
rm -rf cpu/stm32/include/vendor/cmsis/
BOARD=nucleo-l552ze-q make -j4 -C tests/riotboot clean all
```
The issue only manifests if build with multiple processes (which is done on the CI all the time).

#### Expected results
Compilation succeeds without warnings or errors.

#### Actual results
I oftentimes see a git problem like:
```
make: Entering directory 'RIOT/tests/riotboot'
Building application "tests_riotboot" for "nucleo-l552ze-q" with MCU "stm32".

[INFO] cloning stm32cmsis
git-cache: cloning from cache. tag=commitd922865fc0326a102c26211c44b8e42f52c1e53d-33802
compiling RIOT/dist/tools/riotboot_gen_hdr/bin/genhdr...
make: Nothing to be done for 'all'.
[INFO] updating stm32cmsis RIOT/cpu/stm32/include/vendor/cmsis/l5/.pkg-state.git-downloaded
[INFO] updating stm32cmsis RIOT/cpu/stm32/include/vendor/cmsis/l5/.pkg-state.git-downloaded
echo d922865fc0326a102c26211c44b8e42f52c1e53d   > RIOT/cpu/stm32/include/vendor/cmsis/l5/.pkg-state.git-downloaded
[INFO] patch stm32cmsis
echo d922865fc0326a102c26211c44b8e42f52c1e53d   > RIOT/cpu/stm32/include/vendor/cmsis/l5/.pkg-state.git-downloaded
[INFO] patch stm32cmsis
fatal: Unable to create 'RIOT/cpu/stm32/include/vendor/cmsis/l5/.git/index.lock': File exists.

Another git process seems to be running in this repository, e.g.
an editor opened by 'git commit'. Please make sure all processes
are terminated then try again. If it still fails, a git process
may have crashed in this repository earlier:
remove the file manually to continue.
make[1]: *** [RIOT/pkg/pkg.mk:110: RIOT/cpu/stm32/include/vendor/cmsis/l5/.pkg-state.git-patched] Error 128
make: *** [RIOT/cpu/stm32/Makefile.include:66: RIOT/cpu/stm32/include/vendor/cmsis/l5/Include/stm32l5xx.h] Error 2
make: *** Waiting for unfinished jobs....
```

#### Versions
```

Operating System Environment
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 5.9.6-arch1-1 x86_64 unknown
             System shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.18(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 10.2.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 10.2.0
                  avr-gcc: avr-gcc (GCC) 10.2.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 11.0.0

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.7.11
                    cmake: cmake version 3.18.4
                 cppcheck: Cppcheck 2.2
                  doxygen: 1.8.20
                      git: git version 2.29.2
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 3.8.6
                  python2: Python 2.7.18
                  python3: Python 3.8.6
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

aabadie, 2020-11-12:
Do you use ccache ?

cgundogan, 2020-11-12:
>git-cache: cloning from cache. tag=commitd922865fc0326a102c26211c44b8e42f52c1e53d-33802

yup.

kaspar030, 2020-11-12:
I've seen that before: https://github.com/RIOT-OS/RIOT/pull/14152#issuecomment-639555850

kaspar030, 2020-11-12:
I can also trigger this without git-cache: https://github.com/RIOT-OS/RIOT/pull/14152#issuecomment-639771999

IIRC, I could not reproduce when building in Docker, but at the time, docker was not passing -j.

aabadie, 2020-11-12:
> I can also trigger this without git-cache: #14152 (comment)

There's no build failure reported in this comment.

aabadie, 2020-11-12:
I can't reproduce the issue locally (no git-cache, no ccache). What is your version of make ? Mine is 4.3. Murdock workers (on riotbuild) use 4.1.

kaspar030, 2020-11-12:
> There's no build failure reported in this comment.

cmsis is patched twise:

```
[INFO] updating stm32cmsis /home/kaspar/src/riot.tmp/cpu/stm32/include/vendor/cmsis/l0/.pkg-state.git-downloaded
echo 78696f4f79cf51c8b32d0aa5c199a04f57ee4b27   > /home/kaspar/src/riot.tmp/cpu/stm32/include/vendor/cmsis/l0/.pkg-state.git-downloaded
[INFO] patch stm32cmsis
"make" -C /home/kaspar/src/riot.tmp/pkg/stm32cmsis
[INFO] updating stm32cmsis /home/kaspar/src/riot.tmp/cpu/stm32/include/vendor/cmsis/l0/.pkg-state.git-downloaded
echo 78696f4f79cf51c8b32d0aa5c199a04f57ee4b27   > /home/kaspar/src/riot.tmp/cpu/stm32/include/vendor/cmsis/l0/.pkg-state.git-downloaded
[INFO] patch stm32cmsis
```

That succeeds only by chance.

kaspar030, 2020-11-12:
I'm using make v4.3.
Did you try to reproduce the riotbuild example as suggested by @cgundogan? Probably the two parallel compilations break the logic. If the "downloaded" tag file is created *after* downloading, it makes sense that when starting two make instances in parallel, they also start the downloading recipe in parallel (as none of them can finish fast enough for the other to see the already created tag file).

aabadie, 2020-11-12:
> Did you try to reproduce the riotbuild example as suggested by @cgundogan ?

How can I do that on riotbuild ?

I tried what's in the testing procedure but couldn't reproduce locally.

kaspar030, 2020-11-12:
Can you try in Docker?

    $ docker run -ti --rm -v $(pwd):/data/riotbuild -u $(id -u) riot/riotbuild /bin/bash

then

    $ rm -rf cpu/stm32/include/vendor/cmsis/ && BOARD=nucleo-l552ze-q make -j4 -C tests/riotboot clean all

kaspar030, 2020-11-12:
I don't think this is a cmsis issue, more a recipe error for source downloading.

leandrolanzieri, 2021-01-26:
Could you check if #15860 fixes this for you?

PR #15860 (2021-01-27): makefiles/riotboot: ensure BUILDDEPS are built before riotboot targets
### Contribution description
As reported in #15434, there seems to be a race condition when building applications with riotboot with STM32 targets, because of the cloning of CMSIS. This adds `BUILDDEPS` as a dependency for the `riotboot/bootloader/%` targets, so we ensure that all the needed files are there before start building the riotboot binary. This prevents starting the clone of CMSIS twice at the same time, which seems to be the problem reported in #15434.

### Testing procedure
- Follow the steps described in #15434 [... cut]
closed 2021-01-27 by aabadie