RIOT-OS/RIOT #15447: w5100 driver's bug

liyue75, 2020-11-16:
In drivers/w5100/w5100.c, line 186:  
**return (S0_TX_BASE + limit);**,
I think it should be: 
**return (S0_TX_BASE + len - limit);**
It returns the position of the last write.

maribu, 2020-12-18:
Good catch! Yes, indeed. Do you want to create a pull request to fix this, so that you get the credit for the fix?

liyue75, 2020-12-19:
Yes, I do.

leandrolanzieri, 2021-01-07:
Release is approaching, it would be a nice moment to get this fix included :-) @liyue75 ?

liyue75, 2021-01-08:
Hi @leandrolanzieri  That's good to hear! Now I can close this issue.

maribu, 2021-01-08:
There seems to be a misunderstanding. I assumed that you, @liyue75, wanted to open a pull request containing the fix you proposed. So far, the issue is not fixed in master.

Are you interested in opening the pull request? If not, I can open one.

liyue75, 2021-01-08:
@maribu
Er... I was wrong. Now I create a pull request, is it correct?

maribu, 2021-01-08:
Here is [a guide](https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md#contributing-code) on who to contribute code. If you have any questions, feel free to ask here, in the forum, or in the Matrix or IRQ chat room.

Don't feel pushed into this by me - I could also open the pull request if you want. That said, I think this is an excellent opportunity to become [member of the RIOT contributors](https://github.com/RIOT-OS/RIOT/graphs/contributors) ;-)

liyue75, 2021-01-09:
Hi, @maribu. I meet a problem when make a pull request, you can see here,[https://github.com/RIOT-OS/RIOT/pull/15733](url), what is the meaning of "CI: ready for build" label not set?

closed 2021-01-09 by maribu

liyue75, 2021-04-30:
@maribu,
I found there is anoter problem in drivers/w5100/w5100.c, in send function:

static int send(netdev_t *netdev, const iolist_t *iolist)
{
    w5100_t *dev = (w5100_t *)netdev;
    int sum = 0;
    spi_acquire(dev->dev->p.spi, dev->p.cs, SPI_CONF, dev->p.clk);
    uint16_t pos = raddr(dev, S0_TX_WR0, S0_TX_WR1);

    if (pos == 0) {
        pos = S0_TX_BASE;
    }

    for (const iolist_t *iol = iolist; iol; iol = iol->iol_next) {
        size_t len = iol->iol_len;
        pos = tx_upload(dev, pos, iol->iol_base, len);
        sum += len;
    }

    waddr(dev, S0_TX_WR0, S0_TX_WR1, pos);
     ......
should be replaced by

static int send(netdev_t *netdev, const iolist_t *iolist)
{
    w5100_t *dev = (w5100_t *)netdev;
    int sum = 0;
    spi_acquire(dev->dev->p.spi, dev->p.cs, SPI_CONF, dev->p.clk);
    uint16_t tx_wr = raddr(dev, S0_TX_WR0, S0_TX_WR1);
    uint16_t pos = (tx_wr & S0_MASK) + S0_TX_BASE;

    if (pos == 0) {
        pos = S0_TX_BASE;
    }

    for (const iolist_t *iol = iolist; iol; iol = iol->iol_next) {
        size_t len = iol->iol_len;
        pos = tx_upload(dev, pos, iol->iol_base, len);
        sum += len;
    }

    waddr(dev, S0_TX_WR0, S0_TX_WR1, tx_wr + sum);
     ......
Values in TX_WR register should be in range of 0 ~ 0xFFFF, if I don't modify this send function,
it will cause the value of TX_WR to vary between 0 ~ 0x2000.
This bug will cause about 5ms delay every time the send ring buffer is full.