RIOT-OS/RIOT #15457: RIOT OS and Relic Segmentation fault

Stovka, 2020-11-17:
Hello,
I am new to RIOT, I am having issues with Relic library. Compilation works fine. 

#### Description
When I run it, it ends with Segmentation fault on native and kernel panic on real boards. I identified, that the problem is caused by function rand_bytes() in bn_rand(). I would be really thankful if you could help me to solve this problem.

#### Steps to reproduce the issue
make flash term

main.c:
```
#include <stdio.h>
#include "relic.h"

int main(void)
{
    bn_t a, b, c;
	bn_null(a);
	bn_null(b);
	bn_null(c);
	bn_new(a);
	bn_new(b);
	bn_new(c);

	bn_rand(a, BN_POS, RELIC_BN_BITS);
	bn_rand(b, BN_POS, RELIC_BN_BITS);

	bn_add(c, a, b);
	bn_print(c);
	bn_free(a);
	bn_free(b);
	bn_free(c);
	printf("ok");
    return 0;
}
```

Makefile:
```
APPLICATION = relic
BOARD ?= native
RIOTBASE ?= /home/ubuntu/RIOT
export RELIC_CONFIG_FLAGS=-DARCH=ARM -DWORD=32
USEPKG += relic

include $(RIOTBASE)/Makefile.include
```

Relic -> relic_bn_util.c
```
void bn_rand(bn_t a, int sign, int bits) {
	int digits;

	SPLIT(bits, digits, bits, BN_DIG_LOG);
	digits += (bits > 0 ? 1 : 0);

	bn_grow(a, digits);
	printf("rand_bytes\n");
	rand_bytes((uint8_t *)a->dp, digits * sizeof(dig_t));
	printf("rand_bytes");
	a->used = digits;
	a->sign = sign;
	if (bits > 0) {
		dig_t mask = ((dig_t)1 << (dig_t)bits) - 1;
		a->dp[a->used - 1] &= mask;
	}
	bn_trim(a);
}
```
#### Expected results
ok

#### Actual results
main(): This is RIOT! (Version: 2021.01-devel-508-g2f7d6)
make: *** [/home/ubuntu/RIOT/Makefile.include:710: term] Segmentation fault (core dumped)

#### Versions
Windows 10 -> VM Virtual Box -> Ubuntu 20.04.1
```
ubuntu@ubutnu-VirtualBox:~/RIOT$ make print-versions 

Operating System Environment
----------------------------
         Operating System: "Ubuntu" "20.04.1 LTS (Focal Fossa)"
                   Kernel: Linux 5.4.0-53-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: xtensa-esp8266-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: "2.2.0"
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.16.3
                 cppcheck: Cppcheck 1.90
                  doxygen: 1.8.17
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01469-g72a1010c9-dirty (2020-11-06-16:11)
                   python: Python 2.7.18
                  python2: Python 2.7.18
                  python3: Python 3.8.5
                   flake8: 3.7.9 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.8.5 on Linux
               coccinelle: spatch version 1.0.8 compiled with OCaml version 4.08.1

```

Citrullin, 2020-12-19:
Have you checked the [documentation about this](https://github.com/RIOT-OS/RIOT/wiki/Intregate-Relic-(a-library-for-cryptography)-into-RIOT) before? 
Also using ```-DARCH=ARM``` and native isn't a good idea, since your machine probably used x68.
So, you should use ```-DARCH=X86``` instead.

Stovka, 2020-12-21:
1) I tried many combinations (-DARCH=ARM, NONE, x86, x86_64) unfortunately nothing worked.

2) No I havent found that documentation thank you for that! I followed the instructions and now it works. I had to do 2 little changes to make it work:

a) svn checkout doesn't work so I used `git clone https://github.com/relic-toolkit/relic` instead.
b) In riot-build/build.sh I had to change DWORD to WSIZE otherwise it compiles with errors.

relic-build/build.sh:
```                                                                         
#!/bin/bash
ls |grep -v build.sh |xargs rm -r
COMP="-g -m32" LINK="-m32" cmake -DARCH=X86 -DSEED=ZERO -WSIZE=32 -DWITH="DV;MD;BN;CP" -DCMAKE_INSTALL_PREFIX:PATH=/home/ubuntu/RIOT/examples/relic_project/bin/native/usr/local ../relic
```

closed 2020-12-21 by Stovka