RIOT-OS/RIOT #15502: how to run a 6lowpan over ble example?

hguomin8, 2020-11-24:
Hi,

I am new on RIOT-OS, and try to run a 6lowpan over ble sample on my device nrf52840-mdk board by following this guide: https://github.com/RIOT-OS/RIOT/blob/master/pkg/nimble/README.ipv6-over-ble.md, but after I build the application 'examples/gnrc_networking' and flash it to device, type the 'ble info' in device shell, it ouput "shell: command not found: ble", anyone can tell me that how can I config this sample code to run 6lowpan over ble successfully? Thanks.

hguomin8, 2020-11-24:
@leandrolanzieri , can you advise how to make it? much thanks in advance.

leandrolanzieri, 2020-11-25:
Hi @hguomin8. The `ble` command is not being compiled. This seems to be happening because by default in the nrf52840-mdk board, the module `nrf802154` is used, instead of `nimble_netif` (which is the case for other boards, like the nrf52dk). You could try adding by hand the nimble network interface when compiling like so:
```
BOARD=nrf52840-mdk CFLAGS=-DCONFIG_GNRC_IPV6_NIB_SLAAC=1 USEMODULE=nimble_netif make
```
Unfortunately I don't own a nrf52840-mdk to test this out.
I don't have experience with these radios, but maybe @haukepetersen could shed some light on this?

hguomin8, 2020-11-25:
Much thanks @leandrolanzieri , I run the sample successfully by following your suggested compiling options. 

But I also found only one ble device can be connected to the same linux host, this issue is similar with my testing on Zephyr: https://github.com/zephyrproject-rtos/zephyr/issues/30026. did you test muliple 6lowpan over ble connections to the same linux host with your nrf52dk? hi @haukepetersen, may be you also have some insights about this case? 

Thanks for you both!

Citrullin, 2020-12-19:
> But I also found only one ble device can be connected to the same linux host, this issue is similar with my testing on Zephyr: [zephyrproject-rtos/zephyr#30026](https://github.com/zephyrproject-rtos/zephyr/issues/30026). did you test muliple 6lowpan over ble connections to the same linux host with your nrf52dk board? hi @haukepetersen, maybe you also have some insights about this case?

Haven't tested it in a while, but that shouldn't be the case. You can connect multiple nodes. I have done this before.
Try ```echo "connect UU:VV:WW:XX:YY:ZZ 2" >> /sys/kernel/debug/bluetooth/6lowpan_control``` for the seconds, third, fourth one etc.

hguomin8, 2020-12-20:
> Haven't tested it in a while, but that shouldn't be the case. You can connect multiple nodes. I have done this before.
> Try `echo "connect UU:VV:WW:XX:YY:ZZ 2" >> [...]

Thanks @Citrullin, use that command to connect multiple nodes in link layer is ok but the ip level connection of existing connected nodes will be drop once more than one node were connected, please refer to  the issue I open for Zephyr: https://github.com/zephyrproject-rtos/zephyr/issues/30026.

Citrullin, 2020-12-20:
> > > But I also found only one ble device can be connected to the same linux host, this issue is similar with my testing on Zephyr: [zephyrproject-rtos/zephyr#30026](https://github.com/zephyrproject-rtos/zephyr/issues/30026). did you test muliple 6lowpan over ble connections to the same linux host with your nrf52dk board? hi @haukepetersen, maybe you also have some insights about this case?
> > 
> > 
> > Haven't tested it in a while, but that shouldn't be the case. You can connect multiple nodes. I have done this before.
> > Try `echo "connect UU:VV:WW:XX:YY:ZZ 2" >> /sys/kernel/debug/bluetooth/6lowpan_control` for the seconds, third, fourth one etc.
> 
> Thanks @Citrullin, use that command to connect multiple nodes in link layer is ok but the ip level connection of existing connected nodes will be drop once more than one node was connected, please refer to the issue I opened for Zephyr: [zephyrproject-rtos/zephyr#30026](https://github.com/zephyrproject-rtos/zephyr/issues/30026), which is a same issue happened in RIOT.

> Although can't ping each other with global ipv6 address, but it's ok to ping with the link local address:

So, do I get you right? You want to connect two devices to a Linux host and want both of them to be able to route over a global IPv6 address? 
If that is the case, you can't. There isn't RPL routing in Linux. 
There are two ways to fix that.
1. Connect one device and use this one as rpl root and the tree structure behind this one.
2. Use a user-space implementation of RPL in Linux. A couple ones:
- https://github.com/joaopedrotaveira/linux-rpl
- http://unstrung.sandelman.ca/
- https://github.com/tcheneau/simpleRPL

hguomin8, 2020-12-21:
> > > > But I also found only one ble device can be connected to the same linux host, this issue is similar with my testing on Zephyr: [zephyrproject-rtos/zephyr#30026](https://github.com/zephyrproject-rtos/zephyr/issues/30026). did you test muliple 6lowpan over ble connections to the same linux host with your nrf52dk board? hi @haukepetersen, maybe you also have some insights about this case?
> > > 
> > > 
> > > Haven't tested it in a while, but that shouldn't be the case. You can connect multiple nodes. I have done this before.
> > > Try `echo "connect UU:VV:WW:XX:YY:ZZ 2" >> /sys/kernel/debug/bluetooth/6lowpan_control` for the seconds, third, fourth one etc.
> > 
> > 
> > Thanks @Citrullin, use that command to connect multiple nodes in link layer is ok but the ip level connection of existing connected nodes will be drop once more than one node was connected, please refer to the issue I opened for Zephyr: [zephyrproject-rtos/zephyr#30026](https://github.com/zephyrproject-rtos/zephyr/issues/30026), which is a same issue happened in RIOT.
> 
> > Although can't ping each other with global ipv6 address, but it's ok to ping with the link local address:
> 
> So, do I get you right? You want to connect two devices to a Linux host and want both of them to be able to route over a global IPv6 address?
> If that is the case, this has probably something to do with RPL. There isn't RPL routing in Linux.
> There are two ways to fix that.
> 
> 1. Connect one device and use this one as rpl root and the tree structure behind this one.
> 2. Use a user-space implementation of RPL in Linux. A couple ones:
> 
> * https://github.com/joaopedrotaveira/linux-rpl
> * http://unstrung.sandelman.ca/
> * https://github.com/tcheneau/simpleRPL

Actually what I want to do may be simpler than this. I want to connect multiple nodes to a Linux host to form a star network topo, each node just need to communicate with the Linux host with its global IPv6 address, no route between any two nodes needed, and the Linux host act as gateway with a MQTT broker installed and will send the node's data it received from each node to the cloud. In this case, do I still need to implement the userspace rpl?

closed 2022-04-18 by stale[bot]