RIOT-OS/RIOT #15503: api.riot-os.org is resource hungry and makes my fan spin (after Days)

kfessel, 2020-11-24:
#### Description

Story:
I found my laptop fan spinning loudly checked -> found it was Firefox , checked its Task Manger and found api.riot-os.org on top  (this is a place where i would not expect a API documentation page, this should be reserved for marketing and social media garbage)

Diagnose:
api.riot-os.org is resource hungry and  over time the page gets more and more sluggish

what happens:
the showRoot() function in Doxygens navtree.js triggers itself using a timeout after 0 seconds (and keeps doing this) when if fails and it keeps failing due to a missing scrollTo function.

I don't know how it got missing but my guess is it can be fixed with a newer JQuery library.
Why:
The `$()` of the currently used jquerry seems to not forward the scrollTo function. The version that is used on doxygen.org does not have that issue.

#### Steps to reproduce the issue

make the Problem visible:
-With your browser (i tested Firefox and Chromium) visit api.riot-os.org
-Start the inspector tool (ctrl-shift-i)
-performance tab
-record a short time (a second is enough)
   - firefox has a filter for different kind of event where function calls where invisible -> make the visible 
-see masses of function calls to setTimeout in navtree.js:247 showRoot() 

#### Expected results

not masses of function Calls

#### Actual results

masses of function calls to setTimeout in navtree.js:247 showRoot()

#### Versions

current api.riot-os.org build by doxygen 1.8.13 containing JQuerry 1.12.4

kfessel, 2020-11-24:
@kfessel you are wrong there is not update required just add https://github.com/flesler/jquery.scrollTo to Riot

kfessel, 2020-11-24:
The only reason to add this is because jquerry does not forward define scrollTo for its selector thing. I still don't know why.

kfessel, 2020-11-25:
[image]

aabadie, 2020-11-25:
I confirm the issue reported here (tested with firefox). And I also confirm that #15505 is fixing it.

aabadie, 2020-11-25:
Thanks for providing the screenshot because I was not sure what was expected.

PR #15505 (2020-11-25): doc/html: Fix scrollTo issue of doxygen 
Fixes the issue described in #15503

### Contribution description

add a scrollTo for JQuerry since its selector does not forward it

### Testing procedure

Build the doc and use the Performance inspector of your browser to check if the issue described in #15503 is  happening 

see a build version on
https://kfessel.github.io/RIOT/html/

### Issues/PRs references
Fixes: #15503
closed 2020-11-25 by aabadie