RIOT-OS/RIOT #15517: pkg/tinydtls: async event SOCK_ASYNC_CONN_FIN does not indicate which session has been finished

janosbrodbeck, 2020-11-27:
#### Description

When using tinyDTLS with asynchronous sockets the event for a finished session does not indicate which session has been closed. You have 0 information which connection has been finished. It makes it impossible to keep the session for a long time because you cannot track sessions properly without touching the internals of tinydtls.

#### Steps to reproduce the issue

#### Expected results
Argument of the callback event points to the closed session.

#### Actual results
The argument is currently null.
#### Versions

janosbrodbeck, 2020-11-27:
Maybe @pokgak @leandrolanzieri could help?

janosbrodbeck, 2020-11-27:
Changing it to `sock->async_cb(sock, SOCK_ASYNC_CONN_FIN, session);` does work. Of course, it is a `session_t` but this way it's possible to retrieve information about the closed sessions.

pokgak, 2020-11-28:
From the code, `async_cb_arg` is set by the [client][1]/[server][2]: client/server call `sock_dtls_event_init`, which then calls `sock_dtls_set_cb` that [sets `async_cb_arg`][3]. Based on the async tinydtls test, It seems the server does not use the `async_cb_arg` while the client use it to hold data to send later.

I think it makes sense to return a pointer to a `sock_dtls_session_t` for the `SOCK_ASYNC_CONN_FIN` event using the `arg` parameter of the `async_cb`. One thing I'm not sure about is if we're breaking any assumption expected by the handler by not using the `async_cb_arg` given from the client/server.

The way I understand it, it should be okay for sock_dtls to overwrite `async_cb_arg` to pass back data to the client/server. @miri64 is this correct?

[1]: https://github.com/RIOT-OS/RIOT/blob/7237e384368a63e79b3b7273783b5230c29e448b/tests/pkg_tinydtls_sock_async/dtls-client.c#L211-L212
[2]: https://github.com/RIOT-OS/RIOT/blob/7237e384368a63e79b3b7273783b5230c29e448b/tests/pkg_tinydtls_sock_async/dtls-server.c#L169-L170
[3]: https://github.com/RIOT-OS/RIOT/blob/7237e384368a63e79b3b7273783b5230c29e448b/sys/net/sock/async/event/sock_async_event.c#L78

miri64, 2020-11-30:
Yeah, I would set `async_cb_arg` to the session when the event is fired, not when it is handled.

miri64, 2020-11-30:
Mhhh on the other hand, `async_cb_arg` is supposed to be set on callback initialization:

https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/pkg/tinydtls/contrib/sock_dtls.c#L625-L628

Just overriding it might lead to unexpected behavior, so does changing the context for a special event type... Mhhhh.... can't the session be retrieved from the `sock` somehow?

janosbrodbeck, 2020-12-03:
Sorry for the longer delay. 
It does indeed lead to unexpected behaviour. The port was the only correct value in the retrieved session.
```
Endpoint in sock_dtls.c, before cb call: 
Port: 5684
Family: 4 (IPV6)
Address FE:80:00:00:00:00:00:00:C8:0F:4F:FF:FE:13:0E:87

Endpoint in SOCK_ASYNC_CONN_FIN event: 
Port: 5684
Family: 0 (UNDEF)
Address: C0:3A:61:56:01:00:00:00:01:00:00:00:00:D5:A3:F1
```
Hm, I don't see a clean way to hand over the pointer in another way. Of course it is possible to find out which session has been closed by examining the peer list in the tinyDTLS context (which is attached to the socket). But that's inefficient and ugly.

The internal mailbox should not be used for that I guess, especially because then the DTLS event handler must necessarily receive the message. 

For my internal testing in gcoap I just added a function to the DTLS API to retrieve the session. It is stored internally then. You could attach a pointer to the DTLS socket that would be there for exactly this one event, but this is also very unexpected behaviour.

miri64, 2020-12-03:
Mh, do we maybe need another parameter to the callback :-/? But just for that? Seems a bit overkill... Mhhh...

miri64, 2020-12-03:
I think having an extra pointer just for the FIN event in the struct seems like the lesser of two evils in that case.

miri64, 2020-12-03:
There should be a `sock_dtls_get_session()` in any case, IMHO.

janosbrodbeck, 2020-12-03:
true. That could be used for every event to retrieve the session apart from `sock_dtls_receive`. 
I'll look into it!

PR #15755 (2021-01-22): sock/async: add function to retrieve session object of current DTLS event
### Contribution description
This PR adds a new function to sock async: `sock_dtls_get_event_session()` to retrieve the session object within a DTLS event. Currently it is not possible to get this session for every event type (e.g. SOCK_ASYNC_CONN_FIN, SOCK_ASYNC_CONN_RDY). It should work for every possible event type. The issue was described in #15517. 

I didn't get it to work properly with just a pointer in the socket structure, since the session needed to be stored somewhere and storing it was not possible within the [... cut]
closed 2021-01-22 by miri64