RIOT-OS/RIOT #15521: Missing error check of clist_foreach(), mbox_try_put() and msg_try_send()

Hxinrong, 2020-11-30:
Hi,
As shown in the project documentation, function clist_foreach() is used to traverse clist and call function for each member. It returns a node that caused func(node, arg) to exit non-zero, or NULL on empty list or full traversal. But in the following codes, the return value is not checked.
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/drivers/at/at.c#L443

Moreover, at other call sites, the return values of it are checked, for instance,
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/sys/net/gnrc/network_layer/ipv6/ext/frag/gnrc_ipv6_ext_frag.c#L652-L663
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/tests/unittests/tests-core/tests-core-clist.c#L252-L255
****************************************************************************************************************************************
The function mbox_try_put() is used to add message to mailbox, and returns 1 if msg could be delivered, or 0 otherwise. But at the following call sites, I think, there should be some error checks. There is no evidence to prove whether mbox were filled or not.
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/sys/can/conn/raw.c#L123
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/sys/can/conn/raw.c#L207
****************************************************************************************************************************************
Function msg_try_send() has the same problem. It returns 1 if sending a message is successful (message delivered directly or to a queue), 0 if receiver is not waiting or has a full message queue, -1 on error. As follows, there may be a lack of error checks.
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/sys/net/gnrc/network_layer/ipv6/ext/frag/gnrc_ipv6_ext_frag.c#L258
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/sys/net/gnrc/network_layer/ipv6/ext/frag/gnrc_ipv6_ext_frag.c#L269-L271

However, the return values of it from other call sites are checked, for example,
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/sys/net/gnrc/netapi/gnrc_netapi.c#L61-L66

kaspar030, 2020-11-30:
I'll answer the foreach part.
`clist_foreach()` is a generic list traversal tool. Depending on its use, the return value is not relevant.

> clist_foreach(&dev->urc_list, _check_urc, buf);

in this case, a list of "urc" is traversed to find a match using "_check_urc". It is expected to have zero or one matches. So `_check_urc()`, when it finds a match, calls the callback and exits 1, stopping the iteration. The return value of "clist_foreach()" is not needed.

Does that make sense?

@miri64, I'll let you answer the unhandled `msg_try_send()`.

miri64, 2020-11-30:
> @miri64, I'll let you answer the unhandled `msg_try_send()`.

Mh... there could indeed be the potential for a leak there. Will fix!

miri64, 2020-11-30:
@vincent-d could you have a look at the usage of `mbox_try_put()` in `conn_can`, please?

vincent-d, 2020-11-30:
> @vincent-d could you have a look at the usage of `mbox_try_put()` in `conn_can`, please?

I will!

vincent-d, 2020-11-30:
okay, in `conn_can`, those `mbox_try_put()` are called from a xtimer callback, so from interrupt context. There is nothing that can be done on failure. And blocking an interrupt is not an option.
@kaspar030 @miri64 would you agree with this statement?

miri64, 2020-11-30:
Should we maybe add a comment for those cases where errors are not checked?

kaspar030, 2020-11-30:
> Should we maybe add a comment for those cases where errors are not checked?

That would probably save time for people looking at the code. Ideally we'd have some kind of error handling - something more useful (programmatic) than gnrc's "possibly lost interrupt", but even that can be very helpful when debugging.

Hxinrong, 2020-12-01:
> I'll answer the foreach part.
> `clist_foreach()` is a generic list traversal tool. Depending on its use, the return value is not relevant.
>
>
> in this case, a list of "urc" is traversed to find a match [...]

Mh... ok, I see, thank you.

Hxinrong, 2020-12-01:
> Should we maybe add a comment for those cases where errors are not checked?

Well, I think, that would be very useful for people to understand the context. Thanks for your reply.

closed 2020-12-16 by Hxinrong