RIOT-OS/RIOT #15522: Question: Where does the `#include` conding convention come from?

kfessel, 2020-11-30:
Where does the `#include` coding convention come from?

### Description
RIOT-OS has a coding convention of using `#include " "` for its RIOT System includes to differentiate them from `#include < >` which should be used for compiler includes only.

This seems special to me and at least to some other people [PR in ndn-riot](https://github.com/named-data-iot/ndn-riot/pull/14#issuecomment-733788536.) I tried to find the reasoning in the RIOT git history but i failed. In  7caf7881a2eb (14/11/2013) Ludwig Ortmann mentioned the coding convetion (maybe the first time). The Coding_Convetions .md file got added by @maribu  at 25/10/2019 but somehow must have been there before.

As i wrote in the ndn PR at least Arduino and Zephy use `< >` for non compiler library includes.

### Useful links
[a blog post that did some digging through c standard](http://commandlinefanatic.com/cgi-bin/showarticle.cgi?article=art026)
[not very help full gcc manual](http://gcc.gnu.org/onlinedocs/cpp/Search-Path.html)
[microsoft was a little more clear](https://docs.microsoft.com/en-us/cpp/preprocessor/hash-include-directive-c-cpp?view=msvc-160) there are some words about parent includes that if they apply to gcc my lead to some difficult to explain behavior.

###my 2 cent 
Reading the blog-post mentioned above i get the C-Standard does not say anything to distinguish `#include ""` from `#include <>` (i also checked if newer or newer standards say something but the paragraph seems to be unchanged); K&R said something which the RIOT CC conflicts with. 

quoting myself from the ndn PR:
> K&R could have saved us from this discussion if they just had not defined the fallback case for ` " "`
> 
> it would have been you know the path -> `" "`
> compiler has to search -> `< >`

the fallback case for `" "` is what RIOT is using

kaspar030, 2020-11-30:
> Where does the `#include` coding convention come from?

Style. `"foo.h"` is "one of ours", whereas `<foo.h>` is "comes from the runtime/libc/whatever". That only goes upstream, we consider package headers as "ours".
The angle brackets form searches "preconfigured list of standard system directories". For RIOT being a library OS usually compiled to a baremetal target, that means toolchain provided include paths. Quote includes search paths added with `-I` __before those__. So using the quote form __hints to the reader__ that "we don't expect this to be found in the toolchain include paths".

miri64, 2020-11-30:
> The Coding_Convetions .md file got added by @maribu at 25/10/2019 but somehow must have been there before.

For reference, as the [PR importing that .md](https://github.com/RIOT-OS/RIOT/commit/bf117a4e9cbd7607f91f9243dd7554f4a3b6e349#diff-3ca936c6e9f143299296f008dab6650f4914ef24d82715e0011cab9b10863564) indicates: the coding conventions where maintained in the [Wiki](https://github.com/RIOT-OS/RIOT/wiki/Coding-Conventions/feaf3d825e4a86a94179d6ea9f2fe15d40a88521) before. Best way to track the history there is to check out the wiki via git (`git clone git@github.com:RIOT-OS/RIOT.wiki.git` and view the log of the Coding-conventions.md) there.

kfessel, 2020-11-30:
i just checked gcc does not apply what msvc does for the quoted form rule 2 (the parrent include) < this would be a problem 
[incl_test.zip](https://github.com/RIOT-OS/RIOT/files/5616503/incl_test.zip)
```
cd main
gcc -I ../inc1 -I ../inc2 main.c 
```

gcc for quoted -> the currents file directory -> the rest is like angel brackets
gcc for angel   -> search `-I` include paths last added is first to be searched  -> search compiler system includes

the `<>` form also searches -I first before

kfessel, 2022-03-02:
> ... So using the quote form **hints to the reader** that "we don't expect this to be found in the toolchain include paths".

Most other projects use `#include " "` to indicate that the file is found right next to the file including and `< >` to indicat it is in the include path (not caring if it is in the libc or in (lib)<RTOS>)

maribu, 2023-05-17:
Question was answered --> closing this.

closed 2023-05-17 by maribu