RIOT-OS/RIOT #15524: Missing error check of several functions

Hxinrong, 2020-11-30:
Hi,
According to the RIOT API, ds3231_get_time() and ds3231_set_time() are used to get or set date and time from the device, and return 0 on success, or -EIO on I2C communication error. But as shown below, the return values of them have not been checked.
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/tests/driver_ds3231/main.c#L97
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/tests/driver_ds3231/main.c#L125
And in the same file, at the following call sites, they are error checked,
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/tests/driver_ds3231/main.c#L228-L232
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/tests/driver_ds3231/main.c#L221-L225
*****************************************
Function aes_encrypt() is to encrypt one plainBlock-block and save the result in cipherblock. It returns 1 on success, and a negative value if the cipher key cannot be expanded with the AES key schedule. In the following codes, there may be need an error check.
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/sys/random/fortuna/fortuna.c#L69-L72
However, there are error checks from other call sites, for example,
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/tests/sys_crypto/tests-crypto-aes.c#L52-L53
*****************************************
The function cord_common_add_qstring() is used to add selected query string options to a gcoap request, and returns 0 on success, or a negative value on error. As follows, there could be add a check of the return value.
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/sys/net/application_layer/cord/ep/cord_ep.c#L279
What's more, we can see a return value check of it in the fragment below.
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/sys/net/application_layer/cord/epsim/cord_epsim.c#L63-L66
*****************************************
Also, there may be a lack of error checks of function devfs_register(). It is to register a node in DevFS, and returns 0 on success, or a negative value on error. As shown below,
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/boards/mulle/board.c#L177-L178
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/boards/mulle/board.c#L202-L203
Moreover, there is a function test_devfs_register() to test the return values of devfs_register() at the following file. So I think, there will be a need of error checks at the above call sites.
https://github.com/RIOT-OS/RIOT/blob/c337089de58aa5c75395268f483fc5fe11ed3d64/tests/devfs/main.c#L99-L115

kaspar030, 2020-11-30:
@Hxinrong thank's a bunch for reporting! It might make sense to open individual PR's for each test or module, so we can handle assigning them easier.

jeandudey, 2021-03-20:
Linked this issue to the PR #15576 as all of the return codes specified on the issue have been fixed, last one missing is the `aes_encrypt()` one, so when it's merged this one gets closed. (hopefully :-)

closed 2024-01-31 by Teufelchen1