RIOT-OS/RIOT #15538: Crash on context switch on Cortex-M

jimparis, 2020-12-01:
#### Description

The `isr_pendsv` handler in `cpu/cortexm_common/thread_arch.c` uses `r12` as a temporary register, but doesn't save it across a call to `sched_run`.  Per ARM AAPCS, `r12` isn't callee-saved and `sched_run` is free to clobber it.

With my compiler and the nRF52, this makes any optimization level above `-O0` fail with crashes like:

```Context before hardfault:
   r0: 0x00000000
   r1: 0x00000001
   r2: 0x00000000
   r3: 0x20004490
  r12: 0x00000000
   lr: 0x00000111
   pc: 0x00000000
  psr: 0x6000000e

FSR/FAR:
 CFSR: 0x00020000
 HFSR: 0x40000000
 DFSR: 0x00000000
 AFSR: 0x00000000
Misc
EXC_RET: 0xffffffe1
Hard fault occurred in ISR number -2
Attempting to reconstruct state for debugging...
In GDB:
  set $pc=0x0
  frame 0
  bt
```

This was introduced by https://github.com/RIOT-OS/RIOT/pull/14710 @bergzand
and a workaround is:
```diff
diff --git a/cpu/cortexm_common/thread_arch.c b/cpu/cortexm_common/thread_arch.c
index 11ed6a78819e..44b4d3bc92c2 100644
--- a/cpu/cortexm_common/thread_arch.c
+++ b/cpu/cortexm_common/thread_arch.c
@@ -321,7 +321,9 @@ void __attribute__((naked)) __attribute__((used)) isr_pendsv(void) {
     "push   {lr}                      \n" /* push exception return code */
 
     "cpsid  i                         \n" /* Disable IRQs during sched_run */
+    "push   {r12}                     \n" /* save r12 because sched_run may clobber it */
     "bl     sched_run                 \n" /* perform scheduling */
+    "pop    {r12}                     \n" /* restore r12 */
     "cpsie  i                         \n" /* Re-enable interrupts */
```

#### Versions
`arm-none-eabi-gcc (15:8-2019-q3-1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
RIOT 2020.10 (8292037bfe95)

kaspar030, 2020-12-08:
Thanks for reporting @jimparis! I stumbled over this some weeks ago, but thought it was a quirt of newest llvm.
Working on it.
Unfortunately the proposed fix won't work on Cortex-M0.

bergzand, 2020-12-08:
Thanks for reporting, I can look into this later today and adapt the fix to also cover Cortex-M0!

closed 2020-12-08 by bergzand

kaspar030, 2020-12-08:
@jimparis As this was closed automatically, can you confirm that the issue is fixed for you in master?

jimparis, 2020-12-11:
Yes, it's working now.  Thanks!

kaspar030, 2020-12-11:
> Yes, it's working now. Thanks!

Thanks for reporting back!

PR #15635 (2020-12-15): cpu/cortexm_common: fix r12 clobber in pendsv for Cortex-M0
### Contribution description

Previously, isr_pendsv() was relying on r12 not being clobbered by sched_run() which is not correct.

This PR fixes the issue by saving and restoring the scratch register used to save the previous sched_active_thread value, in the partial thumb (Cortex-M0 and related) case.

Some juggling of the r4 (thread content) register is required to make this work.
Testing on the samr21-xpro, for some reason bench_sched_nop is a cycle faster with this change. bench_mutex_pingpong is unaffected.

### Testing procedur [... cut]