RIOT-OS/RIOT #15604: Board native has no CPU_ARCH

kfessel, 2020-12-09:
#### Description

Board native has no CPU_ARCH if one wants to decide something on CPU_ARCH (e.g. which assembler code to compile) its not possible.

#### Steps to reproduce the issue

```
BOARD=adafruit-clue make info-debug-variable-CPU_ARCH
```

```
BOARD=native make info-debug-variable-CPU_ARCH
```

### Expected results

i would expect "x82-32"

#### Actual results

nothing

#### Versions

current master

fjmolinas, 2021-01-20:
Hi @kfessel there isn't really anything currently forcing `CPU_ARCH` to be defined so if  its useful to add this distinction feel free to do so. That  being said I believe kconfig introduced a `CPU_ARCH_NATIVE` as the equivalent so both should be aligned, what you would suggest might be a better value and would adapt this. This would make sense if later a `x82-64` is added for native. @leandrolanzieri what do you think of this?

leandrolanzieri, 2021-01-20:
> @leandrolanzieri what do you think of this?

I agree. The more information the better. A tip for the Kconfig implementation: the macro language allows to call shell commands like so: `$(shell,command)`

fjmolinas, 2021-01-20:
@kfessel do you think you can open a PR for this?

kfessel, 2021-02-01:
> @kfessel do you think you can open a PR for this?

i did but it took me sometime

closed 2022-04-18 by stale[bot]