RIOT-OS/RIOT #15614: problem with iot lab - riot nanocoap server

Eleftheria-99, 2020-12-11:
#### Description

### Useful links

miri64, 2020-12-11:
Hi @Eleftheria-99, welcome to the riot community! The `microcoap` example was moved to tests in 75bb211c8bf48a0c8a14a8f8edcc56329061efb5, so please use `tests/pkg_microcoap` instead of `examples/microcoap_server`. @aabadie @fjmolinas can these tutorials be updated somehow?

aabadie, 2020-12-11:
> can the IoT-LAB tutorials be updated somehow?

Yes, anyone can open a PR on the [website repo](https://github.com/iot-lab/iot-lab.github.io).

miri64, 2020-12-11:
@Eleftheria-99 did that fix your issues?

Eleftheria-99, 2020-12-11:
I found out they have changed the name of the directory from micro coap to
nanocaop_server and that was it was not working ! I did cd ~/riot/RIOT and
then ls and then I saw it !!

aabadie, 2020-12-11:
@Eleftheria-99 instead of `examples/microcoap`, you should use `examples/nanocoap_server`. This is a mistake in the tutorial because the version of RIOT was updated to 2020.04 where `microcoap` is not located in the examples anymore. Can you try the tutorial with `examples/nanocoap_server` instead ?
If it works, the tutorial will have to be adapted. As I said this is as simple as opening a PR on [website repo](https://github.com/iot-lab/iot-lab.github.io). Thanks!

miri64, 2020-12-11:
Great! Be aware, that `nanocoap` is a different CoAP implementation than `microcoap`.

miri64, 2020-12-11:
But your problem seems to be fixed, so I'm closing this issue.

closed 2020-12-11 by miri64

Eleftheria-99, 2020-12-11:
Yes , I am looking into the difference between micro coap and nano coap,
although in the tutorial they are being mentioned both.
( Steps 6-11 ). I think they have mixed something!

aabadie, 2020-12-11:
> I think they have mixed something!

The CoAP tutorial version for A8-M3 is already using `nanocoap_server` and the commands are correct so I think I just forgot to also update the command lines in the M3 version...
I'm the author of these tutorials, so you can blame me :)

Eleftheria-99, 2020-12-11:
It is okay , I just lost 2 hours trying to find why it does not work, but
you should update the tutorial in order others no to have the same problem!
Also, I tried the MQTT-SN tutorial and in the 10th step I need to use the
third node M3,  but the experiment contains only 2 A8 nodes. So, should I
create a M3 node for the experiment? If yes , how can I add the 3rd node ?

Thank you

fjmolinas, 2021-01-13:
> It is okay , I just lost 2 hours trying to find why it does not work, but you should update the tutorial in order others no to have the same problem! Also, I tried the MQTT-SN tutorial and in the [...]

ping @aabadie!