RIOT-OS/RIOT #15615: pkg/tinydtls: Question about the design of sock_dtls_session_t

janosbrodbeck, 2020-12-11:
#### Description
I have a few questsions regarding the implementation of `sock_dtls_session_t` of the tinyDTLS implementation. This is the struct for tinyDTLS:
```
struct sock_dtls_session {
 sock_udp_ep_t ep; 
 session_t dtls_session;
};
```
It consists of a UDP endpoint and a session_t struct, which itself is a session object of tinyDTLS. But if you look at `session_t`
```
typedef struct {
 unsigned char size;
 ipv6_addr_t addr;
 unsigned short port;
 int ifindex;
} session_t;
```
you can see, that it is just an UDP endpoint only represented differently. So...a `sock_dtls_session_t` for tinyDTLS is a UDP endpoint...twice. 

If I run into problems with the DTLS API I came to the conclusion that I am mostly bogging around the problem that sometimes one of the endpoints is not properly set. Additionally, a session object costs really much memory in its current state. So storing this struct should be highly avoided IMHO. 

Is there a reason for this double UDP endpoint design? As far as I can tell the `session_t` should be sufficient?

I can imagine that the idea behind this structure was, that you can easily set the session endpoint from a `sock_udp_ep_t` and that the GNRC udp endpoint is available for every DTLS implementation of `sock_dtls_session_t`. But I would always prefer an official way to set and retrieve the UDP endpoint from a session, so that I don't care how the endpoint is stored internally:
```
sock_dtls_session_get_udp_ep(...)
sock_dtls_session_set_udp_ep(...)
```

janosbrodbeck, 2020-12-11:
Maybe @pokgak @leandrolanzieri have some insight here?

miri64, 2020-12-11:
I don't remember exactly, but my guess is that the enp-point is in the session since it can't be guaranteed that the address and port are always in the session struct (which is implementation specific to tinyDTLS in your case). The other alternative could be that one represents the local end-point of a session, the other the remote.

janosbrodbeck, 2020-12-11:
They are equal since there are some helper functions `_session_to_ep`, `_ep_to_session` for that. They only copy the endpoint from one struct to the other, and they are called quite frequently to guarantee that both are set.

`it can't be guaranteed that the address and port are always in the session struct`
But the address and port have to be set before using the session anyway. Of course it makes it easier to set, because you can just set the GNRC udp endpoint in the session from your GNRC udp endpoint...but I would prefer a generic way for that anyway, since you cannot guarantee that this struct exists for every DTLS implementation. So setting it via functions like
```
sock_dtls_session_set_udp_ep
sock_dtls_session_get_udp_ep
```

janosbrodbeck, 2020-12-11:
Oh, have misunderstood you.

If the the implementation specific session struct does not contain the udp endpoint I would say that the UDP endpoint then needs to be added to `sock_dtls_session_t` to ensure that the session contains port, address. But `sock_dtls_session_t` is also implementation specific to address such problems.

pokgak, 2020-12-11:
It's also been a while for me and honestly at the time, I'm just fumbling around trying to come up with an acceptable API. The only thing I can remember is in my earlier implementation, there used to be a problem where either one of `ep->netif` or `session->ifindex` is 0 (not set) after the handshake. I can't remember why but that seems to be fixed now.

Here is a comparison of the `session_t` and `sock_udp_ep_t`:
```C
struct _sock_tl_ep {
    int family;
    union {
        uint8_t ipv6[16];
        uint8_t ipv4[4];        /**< IPv4 address mode */
        uint32_t ipv4_u32;      /**< IPv4 address *in network byte order* */
    } addr;                 /**< address */

    uint16_t netif;
    uint16_t port;          /**< transport layer port (in host byte order) */
};

/* session_t */
typedef struct {
    unsigned char size;
    ipv6_addr_t addr;
    unsigned short port;
    int ifindex;
} session_t;
```

Essentially, they contain the same information. Considering that we already convert ep to session all the timer currently, removing `sock_udp_ep_t` from `sock_dtls_session_t` could save us some memory as you said. `sock_dtls_session_t` is also implementation specific, as long as changing that does not require changing the API, I'm not against it.

miri64, 2020-12-11:
> […] as long as changing that does not require changing the API, I'm not against it.

I would at least like to have the API additions @janosbrodbeck proposed, to get the end-point from a session. Not sure the setter is needed, though.

janosbrodbeck, 2020-12-11:
Setter is needed/useful because before using a session the first time you have to set the endpoint (in whatever way it is represented in the session). Having a generic way would be my preferred way, because otherwise I would need to look into every DTLS implementation how to prepare a session.

I'll do the API addition, at least the getter for now.  But even then: would you expect `sock_dtls_session_get` to retrieve the `sock_udp_ep_t` endpoint or the `session_t` ep? I dare to say that both are valid, and dare to say that `session_t` is the one that is/was ultimately also worked with internally.

I also try out on a branch to remove the `sock_udp_ep_t`. I currently don't see any changes for the API or the functionality of the API if we remove it. If it is possible and it comes to a PR we can still talk about it in detail I guess and drop it if something really speaks against it. At the moment it is really just an assumption that the second endpoint is not needed. 
(But I'll have to see when I get around to it)

miri64, 2020-12-11:
> Setter is needed/useful because before using a session the first time you have to set the endpoint (in whatever way it is represented in the session). Having a generic way would be my preferred way, because otherwise I would need to look into every DTLS implementation how to prepare a session.

Ah, well then there actually _has_ to be a setter, not just a preferred way. Otherwise the API is not implementation-agnostic.

> I'll do the API addition, at least the getter for now. But even for that: would you expect `sock_dtls_session_get` to retrieve the `sock_udp_ep_t` endpoint or the `session_t` ep (converted to sock_udp_...)? I dare to say that both are valid, and dare to say that `session_t` is the one that is/was ultimately worked with internally.

Maybe I misunderstood, but I thought the idea was to remove the `sock_udp_ep_t` from the session altogether?

miri64, 2020-12-11:
If they differ one of them must be wrong, I don't understand how they could both be valid.

janosbrodbeck, 2020-12-11:
> Maybe I misunderstood, but I thought the idea was to remove the sock_udp_ep_t from the session altogether?

Even in its current state, I would never rely on any DTLS implementation to explicitly include the `sock_udp_ep_t`.
Provide getter and setter and the user of the API does not have to fuzzle around the internals.

>  If they differ one of them must be wrong, I don't understand how they could both be valid.

Basically it is currently this way if you use a session (e.g. for the first time, wanting to do a handshake):
- Set session.ep to the remote endpoint you do want to make a handshake with
- Internally: session.dtls_session gets set from the session.ep in `sock_dtls_session_init`
- tinydtls uses the session.dtls_session

If you retrieve a session via `sock_dtls_recv`:
- Internally: tinyDTLS provides the dtls_session and the session.ep gets set from the dtls_session
- Event occurs with `sock_dtls_session_t` with session.ep and session.dtls_session set

Okay yes you would return session.ep to save another conversion.

janosbrodbeck, 2020-12-11:
Sometimes it would be handy to already have another implementation to at least have a clue how a session object would look like there :sweat_smile: