RIOT-OS/RIOT #15623: ICMPv6: no response to fragmented echo request messages from RIOT

benpicco, 2020-12-13:
#### Description

#### Steps to reproduce the issue

This came up while testing #15572, but the issue is also reproducible on `native`.

```
# create a TAP interface
sudo ip tuntap add tap0 mode tap user ${USER}
sudo ip link set tap0 up

# start native with IPv6 fragmentation enabled
USEMODULE=gnrc_ipv6_ext_frag make -j -C examples/gnrc_networking BOARD=native all term

# we can ping the link-local address of the tap0 interface with fragmented packets
> ping fe80::605d:7cff:fe2b:c4fc -s 2000
ping fe80::605d:7cff:fe2b:c4fc -s 2000
2008 bytes from fe80::605d:7cff:fe2b:c4fc%6: icmp_seq=0 ttl=64 time=0.269 ms
2008 bytes from fe80::605d:7cff:fe2b:c4fc%6: icmp_seq=1 ttl=64 time=0.211 ms
2008 bytes from fe80::605d:7cff:fe2b:c4fc%6: icmp_seq=2 ttl=64 time=0.198 ms

--- fe80::605d:7cff:fe2b:c4fc PING statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 0.198/0.226/0.269 ms
```

We can ping the `native` instance from Linux using non-fragmented packets

```
% ping fe80::605d:7cff:fe2b:c4fd%tap0 -c 3
PING fe80::605d:7cff:fe2b:c4fd%tap0(fe80::605d:7cff:fe2b:c4fd%tap0) 56 data bytes
64 bytes from fe80::605d:7cff:fe2b:c4fd%tap0: icmp_seq=1 ttl=64 time=0.224 ms
64 bytes from fe80::605d:7cff:fe2b:c4fd%tap0: icmp_seq=2 ttl=64 time=0.196 ms
64 bytes from fe80::605d:7cff:fe2b:c4fd%tap0: icmp_seq=3 ttl=64 time=0.165 ms

--- fe80::605d:7cff:fe2b:c4fd%tap0 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2035ms
rtt min/avg/max/mdev = 0.165/0.195/0.224/0.024 ms
```

#### Expected results

We can also ping the native instance using fragmented packets.

#### Actual results

```
% ping fe80::605d:7cff:fe2b:c4fd%tap0 -c 3 -s 2000
PING fe80::605d:7cff:fe2b:c4fd%tap0(fe80::605d:7cff:fe2b:c4fd%tap0) 2000 data bytes

--- fe80::605d:7cff:fe2b:c4fd%tap0 ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2038ms
```

reception of fragmented packets does work though

```
# send 2k payload to the native instance
dd if=/dev/zero of=/tmp/2k.raw count=4
nc -u fe80::605d:7cff:fe2b:c4fd%tap0 1234 < /tmp/2k.raw

# it is received on the other side
PKTDUMP: data received:
~~ SNIP  0 - size: 2048 byte, type: NETTYPE_UNDEF (0)
00000000  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00
…
000007F0  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00
~~ SNIP  1 - size:   8 byte, type: NETTYPE_UDP (4)
   src-port: 51738  dst-port:  1234
   length: 2056  cksum: 0xe2e3
~~ SNIP  2 - size:  40 byte, type: NETTYPE_IPV6 (1)
traffic class: 0x00 (ECN: 0x0, DSCP: 0x00)
flow label: 0x61b51
length: 2056  next header: 17  hop limit: 64
source address: fe80::605d:7cff:fe2b:c4fc
destination address: fe80::605d:7cff:fe2b:c4fd
~~ SNIP  3 - size:  20 byte, type: NETTYPE_NETIF (-1)
if_pid: 6  rssi: 0  lqi: 0
flags: 0x0
src_l2addr: 62:5D:7C:2B:C4:FC
dst_l2addr: 62:5D:7C:2B:C4:FD
~~ PKT    -  4 snips, total size: 2116 byte
```

#### Versions
RIOT master

miri64, 2020-12-14:
Can you confirm that your testing procedures are basically identical to the [release specs 3.6](https://github.com/RIOT-OS/Release-Specs/tree/master/03-single-hop-ipv6-icmp#task-06---icmpv6-link-local-echo-on-native-ipv6-fragmentation)? Because those just [passed last Saturday](https://github.com/RIOT-OS/RIOT/runs/1541015288?check_suite_focus=true#step:9:43). So this seems to be more of a configuration issue.

miri64, 2020-12-14:
From re-reading https://github.com/RIOT-OS/Release-Specs/issues/137, it should be enough to increase the packet buffer size to 8192 when using such large packets. This is the same that we did for the test application that we use in 3.6:

https://github.com/RIOT-OS/RIOT/blob/cc55d4c0a6a8e9e8e144b09cfd39a7ed4c62fe44/tests/gnrc_udp/Makefile#L21-L24

benpicco, 2020-12-14:
Ah yes, I already suspected something like this.

```
% ping fe80::6422:f8ff:fe28:4245%tap0 -s 2000
PING fe80::6422:f8ff:fe28:4245%tap0(fe80::6422:f8ff:fe28:4245%tap0) 2000 data bytes
2008 bytes from fe80::6422:f8ff:fe28:4245%tap0: icmp_seq=1 ttl=64 time=0.304 ms
2008 bytes from fe80::6422:f8ff:fe28:4245%tap0: icmp_seq=2 ttl=64 time=0.312 ms
2008 bytes from fe80::6422:f8ff:fe28:4245%tap0: icmp_seq=3 ttl=64 time=0.324 ms
^C
--- fe80::6422:f8ff:fe28:4245%tap0 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2034ms
rtt min/avg/max/mdev = 0.304/0.313/0.324/0.008 ms
```

closed 2020-12-14 by benpicco

dylad, 2020-12-14:
Good to know, thanks for the tips !